open62541/open62541 #7025: OPCUA Client and encryption authentification

lesslefou, 2025-01-15:
## Description
Hi, 
I am actually working on creating an OPCUA Client in order to used it with my PLC. 
I was able to configure a connection with no user and no security policy for the moment. 

I now want to create a method which allows only a connection with an encryption mode and a certificate. 

I try a lot of thing without success for the moment. I think i miss differents parts by not understanding everything. 
I was trying to connect my client to the prosys OPCU Simulation server. I got the following error : 
[image]
In the certificate tab of my server, i din't see the certificate in order to trust it before attempting a new connection. 

Is there anyone who can help me with this kind of authentification ? 

Thank you for your precious help. 

## Background Information / Reproduction Steps
This is my last attempt code on the subject. 
`
    UA_StatusCode retval = UA_STATUSCODE_BAD;

    if (m_client != nullptr) {
        UA_Client_delete(m_client);
        m_client = nullptr;
    }

    //Create new client
    m_client = UA_Client_new(); 

    //Load certificate and private key
    UA_ByteString certificate = loadFile(m_certificatePath.c_str());
    UA_ByteString privateKey  = loadFile(m_privateKeyPath.c_str());
    
    //Set configuration
    UA_ClientConfig *config = UA_Client_getConfig(m_client);
    config->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
    UA_CertificateVerification_AcceptAll(&config->certificateVerification);
    config->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256");
    config->applicationUri = UA_String_fromChars("urn:TLN-LDU-001:ProsysOPC:UaBrowser");
    retval = UA_ClientConfig_setDefaultEncryption(config, certificate, privateKey, NULL, 0, NULL, 0);
    if(retval != UA_STATUSCODE_GOOD) {
        ROS_ERROR_STREAM("LinkOpcua::connectWithEncryption : failed to set encryption");
        UA_Client_delete(m_client);
        m_client = nullptr;
        return false;
    }

    //Connect the client
    retval = UA_Client_connect(m_client, m_endpointUrl.c_str());
    
    `

Used CMake options:

```bash
cmake -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_ENCRYPTION=OPENSSL ..

```

jpfr, 2025-01-16:
Hi, which version of open62541 do you use?

lesslefou, 2025-01-22:
Hi,

I'm not sure but the AppUri should appear on the certificat information ?

For example I have generated my own certificate like this:

`X509v3 Key Usage:
Digital Signature, Non Repudiation, Key Encipherment, Data Encipherment, Certificate Sign

X509v3 Extended Key Usage:
TLS Web Server Authentication, TLS Web Client Authentication

X509v3 Subject Alternative Name:
DNS:myhostname.mydomain.local, URI:urn:myhostname.mydomain.local:OPCUA:SimulationServer
`

How to define correctly the AppUri in the certificat ? Should be include in the URI:urn ....

jpfr, 2025-01-22:
The ApplicationUri from the configured ApplicationDescription goes into the Subject Alt Name.

Our generation script takes it as a command line argument:
https://github.com/open62541/open62541/blob/master/tools/certs/create_self-signed.py

lesslefou, 2025-01-22:
after changing the applicationUri by the correct URI on my code i got these rejecting endpoints : 
[image]

lesslefou, 2025-01-22:
I don't really understand. I prefere to use my software to create the certificate because i don't want to use a self-signed certificat and in your python code I don't really see wath I need to change in my certificate. 

With the code I copy on the issue message and the detail of my certificate in my previous message, what did i need to change in order to get the message (cf issus) go to "Connected" ?

Is any fields in my certificate wrong ?

lesslefou, 2025-01-22:
Ok I will try this. Thank you for your help

lesslefou, 2025-01-30:
@electro-logic I try to create a certificate with the URL parameter but I got the error that URL is an unsupported option. 

How can I do that ?

This is my try : 

**v3.ini**
authorityKeyIdentifier=keyid,issuer
basicConstraints=CA:FALSE
keyUsage = digitalSignature, keyEncipherment
extendedKeyUsage = serverAuth, clientAuth
subjectAltName = @alt_names

[ alt_names ]

URL.1 = urn:myhostname:OPCUA:MyApp
DNS.1 = myhostname

**script.sh**
//sign the certificate with the rootac
openssl x509 -req -in myhostname.csr -CA rootac.crt -CAkey rootac.key -CAcreateserial \
-out myhostname.crt -days 365 -sha256 -extfile v3.ini

lesslefou, 2025-01-31:
the method you suggest is using an URL:urn and not URI:urn. I don't understand your previous message in this case.. 

Do you have a certificate which works that I can use to test my code and analyse ?

lesslefou, 2025-01-31:
how can i do that ? sorry I'm really lost in all this encryption stuff...

lesslefou, 2025-02-03:
Update : I now have a correct certificate and private key for both my server and my client. I am able to connect with my simulation app (Prosys UA Browser for the client part and Prosys UA Simulation Server for the server). 

I try to make it work with my code but failed.. 

Here is my last version of code + debug error I got : 

`
//Create new client
    m_client = UA_Client_new(); 

    /* Load certificate and private key */
    UA_ByteString certificate = loadFile(m_certificatePath.c_str());
    UA_ByteString privateKey  = loadFile(m_privateKeyPath.c_str());
    
    //Set configuration
    UA_ClientConfig *config = UA_Client_getConfig(m_client);

    config->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
    config->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
    config->applicationUri = UA_String_fromChars("urn:simulation.application");
    config->clientDescription.applicationUri = UA_String_fromChars("urn:simulation.application");
    
    retval = UA_ClientConfig_setDefaultEncryption(config, certificate, privateKey, NULL, 0, NULL, 0);   
    if(retval != UA_STATUSCODE_GOOD) {
        return false;
    }

    UA_CertificateVerification_AcceptAll(&config->certificateVerification);  

    //Connect the client
    retval = UA_Client_connect(m_client, m_endpointUrl.c_str());
`

[image]

I understand why the Application URI is skip for the non security policy. But What I don't understand is why the connection is made with this one and not "Basic256Sha256" which is the one I ask in my code.

NoelGraf, 2025-02-14:
The issue seems to be resolved. Please reopen if needed.

closed 2025-02-14 by NoelGraf