open62541/open62541 #7034: failed to build 1.4.9 on macos

chenrui333, 2025-01-21:
👋 seeing some build issue with 1.4.9 release as 

```
  /tmp/open62541-20250120-4734-s2hp9n/open62541-1.4.9/plugins/ua_log_stdout.c:49:19: error: use of undeclared identifier 'PTHREAD_RECURSIVE_MUTEX_INITIALIZER_NP'
     49 | UA_Lock logLock = UA_LOCK_STATIC_INIT;
        |                   ^
  /tmp/open62541-20250120-4734-s2hp9n/open62541-1.4.9/build/src_generated/open62541/config.h:404:30: note: expanded from macro 'UA_LOCK_STATIC_INIT'
    404 | #define UA_LOCK_STATIC_INIT {PTHREAD_RECURSIVE_MUTEX_INITIALIZER_NP, 0}
        |                              ^
  1 error generated.
```

and this is my build patch, https://github.com/open62541/open62541/commit/ddff3a1bd33ccbda6456d9ae2c2d408ea718b47b

relates to https://github.com/Homebrew/homebrew-core/pull/204972

jpfr, 2025-01-22:
Hey @chenrui333,

right. It would be great if you could submit a PR.
You should use PTHREAD_RECURSIVE_MUTEX_INITIALIZER on OSX.

jpfr, 2025-01-23:
Closed via #7042

closed 2025-01-23 by jpfr