open62541/open62541 #7036: Async methods without threads

martinhaefner, 2025-01-21:
With the new event loop interface of version 1.4, it is possible to handle all I/O within the 'open62541' thread by just adding own FDs to an externally connected e.g. POSIX event loop. 

Now, on incoming method calls, it would be possible to issue e.g. an I/O bound operation and wait for the response on a certain FD via the main event loop. On arrival of the response, the response handler could again be evaluated within the main thread and a method call response could be returned to the caller without any threads and locks involved. Even with other application threads involved, the method call handling could be single-threaded only, leaving the complexity of the thread handling to the application programmer. 

What I want to say is, it seems the open62541 API would already be quite fine for 'asynchronous in-thread' handling of method requests. Some code depends on threading macros, but the main problem is that the library always expects the output parameters being set within the callback instead of providing a separate library call for returning the response parameters:

<pre>
// synchronous case
void my_method_callback(UA_Server*, UA_Variant* inputArgs)
{
   OutputArgs args[];
   ...
   UA_Server_setupResponse(UA_STATUSCODE_GOOD, outputArgs);
}

// asynchronous case
void my_method_callback(UA_Server*, UA_Variant* inputArgs)
{
   my_response_handle = UA_Server_deferResponse(...);
}

void any_fd_callout_from_eventloop()
{
   UA_Variant outputArgs[];
   UA_Server_setupResponse(my_response_handle, UA_STATUSCODE_GOOD, outputArgs);
}
</pre>

Inspecting the example, one could delay the provision of the output arguments after the retrieval of the UA_CallbackResult object from the `UA_Server_call()` and before returning the response via `UA_Server_setAsyncOperationResult()`, but this looks like a crazy hack to me.

Did you ever think about such an async in-thread API implementation?

jpfr, 2025-01-22:
Our current approach for async methods assumes there are "worker threads".
But yes, you can also implement this worker thread instead as integrated with the EventLoop in a single thread.

We are thinking of simplifying the async API as follows.

- We define a special statuscode UA_STATUSCODE_INTERNAL_ASYNCHRONOUS
- When the "normal" user-defined method callback returns this status code, then we "freeze" the response for the asynchronous operation until it gets finalized.
- We define a method `UA_Server_finalizeOperation` which the user-defined code calls after an async operation has finished.

That way we no longer require any special definitions in the MethodNode.
And we can use the same also for read/write.

Please verify that the above would be compatible with the usage scenario you have mind.

martinhaefner, 2025-01-22:
Yes, that's exactly what I was thinking about.
Do I get some kind of handle in order to identify the operation I would
like to finalize, since multiple operations could be pending
simultaneously, or what would the `UA_Server_finalizeOperation` Signature
look like?

jpfr, 2025-01-22:
The hande could be just the pointer to the target datastructure.
But it must be unique. Also when an async operations times out...
We'll make a proposal.