open62541/open62541 #7043: "The server certificate is different from the EndpointDescription" been reported even when No security profile is used( NONE)

Graos, 2025-01-24:
**OPEN62541 OPC UA STACK VERSION : v1.4.8-6-g0cc73feb0**

Steps to reproduce:

Start an Open62541 pubSub Server + Client and connect to an third party OPC UA server using NONE security Policy.
Connection is successful between Open62541 node and OPC UA server and after about three minutes the connection is closed by Open62541 node with error "The server certificate is different from the EndpointDescription" processing the message returned the error code BadSecurityChecksFailed.

Log file from open62541 node attached.

Question is
1. Why it accepts connection initially and then closes it after three minutes with the above message ?
2. why certificate is validated with endpoint description when NONE security policy is used ?

Log file attached.

[log.txt](https://github.com/user-attachments/files/18534324/log.txt)

GoetzGoerisch, 2025-01-24:
Should be fixed by https://github.com/open62541/open62541/pull/6955 - Contained in 1.4.9

jpfr, 2025-02-08:
Please reopen if still not solved.

closed 2025-02-08 by jpfr