open62541/open62541 #7056: open62541.c,.h files not generated per procedure "Building a Simple Server"

Brian12412, 2025-01-28:
## Description
Followed "Building with Visual Studio on Windows" per https://www.open62541.org/doc/master/building.html
Followed "Building a Simple Server" per https://www.open62541.org/doc/master/tutorial_server_firststeps.html
Selected “amalgamation” per above in Visual Studio 2022 CMake folder/project
Successfully performed "Build All"
open62541.c,.h files were not generated
## Background Information / Reproduction Steps
See above

Used CMake options:
UA_ENABLE_AMALGAMATION checked
No other changes
## Checklist

Please provide the following information:
Information provided for all applicable items below
 - [ x] open62541 Version (release number or git tag): 1.4.9 Latest
 - [ x] Other OPC UA SDKs used (client or server):  none
 - [ x] Operating system: Microsoft Windows 10 Enterprise, Build 19045
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Brian12412, 2025-01-28:
Additional Information: Verified that file "CMakeCache.txt" contains the following:
//Concatenate the library to a single file open62541.h/.c
UA_ENABLE_AMALGAMATION:BOOL=True

jpfr, 2025-01-29:
Fixed via #7054.
Before that you might need a dedicated `make open62541-amalgamation`.

closed 2025-01-29 by jpfr