open62541/open62541 #7061: Client loop returns UA_STATUSCODE_BADTIMEOUT

hmaiedEtic, 2025-01-29:
## Description

Hello,

I am currently using the client loop example `(client_connect_loop.c`) to connect to the Unified Automation Server. The client successfully establishes a connection, but it is returning the following error: `UA_STATUSCODE_BADTIMEOUT`. I have the same  problem with open62541 server.
    
```
[2025-01-29 11:27:41.138 (UTC+0100)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2025-01-29 11:27:41.139 (UTC+0100)] info/channel	Connection 3 | SecureChannel 1506711796 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2025-01-29 11:27:41.139 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-01-29 11:27:41.140 (UTC+0100)] info/client	The initially defined EndpointURL opc.tcp://192.168.0.35:48022is valid for the server
[2025-01-29 11:27:42.138 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:43.139 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:44.045 (UTC+0100)] warn/client	The server returned Endpoints with a different EndpointUrl opc.tcp://192.168.0.35:48022 than was used to initialize the connection: . Some servers require a complete match of the EndpointUrl/DiscoveryUrl (including the path) to return all endpoints.
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	Rejecting endpoint 0: security policy not available
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	Rejecting endpoint 1: security policy not available
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	Rejecting endpoint 2: security policy not available
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	Rejecting endpoint 3: security policy not available
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	Rejecting endpoint 4: security policy not available
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	Selected endpoint 5 in URL opc.tcp://192.168.0.35:48022 with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	Selected UserTokenPolicy Anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-01-29 11:27:44.045 (UTC+0100)] info/client	The initially defined EndpointURL opc.tcp://192.168.0.35:48022is valid for the server
[2025-01-29 11:27:44.140 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:45.141 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:46.140 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:47.141 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:48.142 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:49.143 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:50.143 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:51.144 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:52.145 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:53.145 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:53.146 (UTC+0100)] info/client	The initially defined EndpointURL opc.tcp://192.168.0.35:48022is valid for the server
[2025-01-29 11:27:53.146 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2025-01-29 11:27:53.149 (UTC+0100)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2025-01-29 11:27:53.151 (UTC+0100)] info/userland	date is: 29-01-2025 10:27:52.963
[2025-01-29 11:27:54.152 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:55.152 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:56.153 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:57.153 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:27:58.153 (UTC+0100)] warn/client	Client already connected
[2025-01-29 11:27:59.153 (UTC+0100)] error/client	Not connected. Retrying to connect in 1 second
[2025-01-29 11:28:00.154 (UTC+0100)] warn/client	Client already connected
```

Has anyone experienced this issue or could suggest potential causes ?

Thanks in advance for your help!

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.9
 - [x] Other OPC UA SDKs used (client or server): UAcppServer(unified automation)
 - [x] Operating system: ubuntu-22.04.4
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

hmaiedEtic, 2025-01-29:
I have the same problem in v1.3.15 but not in v1.2

jpfr, 2025-02-05:
Ok. We will try to reproduce.
Any experience with our 1.4 branch so far?

hmaiedEtic, 2025-03-03:
Any update please ?