open62541/open62541 #7062: Cannot connect to endpoint with SecurityPolicy#None returns BadIdentityTokenRejected

RSpeets, 2025-01-30:
I try to connect to a MatrikonOpcUaWrapper that has only one enpoint, but the client refuses.
Earlier versions of open62541 (v1.3.1) were able to connect. Also other OPCUA clients have no problem connecting to this endpoint anonymously.

After some digging it seems to be an incorrect server setting in the user identity setup. I logged the endpoints and user identities:

Endpoint[0]:
Endpoint URL: opc.tcp://[hidden]/MatrikonOpcUaWrapper
Transport profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
Security Mode: 1 None
Security profile URI: http://opcfoundation.org/UA/SecurityPolicy#None
Security Level: 0
> User Identity[0]:
  issuedTokenType:
  issuerEndpointUrl:
  policyId: Anonymous
  securityPolicyUri: **http://opcfoundation.org/UA/SecurityPolicy#Basic256**
  tokenType: 0

Please provide the following information:

 - [x ] open62541 Version (release number or git tag): 1.4.9
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

[2025-01-30 07:49:34.749 (UTC+0100)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-01-30 07:49:34.749 (UTC+0100)] info/network       TCP 948 | Opening a connection to "10.104.174.90" on port 21381
[2025-01-30 07:49:34.764 (UTC+0100)] info/channel       TCP 948 | SC 185        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.0s
[2025-01-30 07:49:34.764 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-01-30 07:49:34.764 (UTC+0100)] info/client        Use the EndpointURL opc.tcp://[hidden]:21381/MatrikonOpcUaWrapper returned from FindServers and reconnect
[2025-01-30 07:49:34.764 (UTC+0100)] info/channel       TCP 948 | SC 185        | SecureChannel closed
[2025-01-30 07:49:34.764 (UTC+0100)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-01-30 07:49:34.764 (UTC+0100)] info/network       TCP 964 | Opening a connection to "10.104.174.90" on port 21381
[2025-01-30 07:49:34.764 (UTC+0100)] info/network       TCP 948 | Socket closed
[2025-01-30 07:49:34.764 (UTC+0100)] info/channel       TCP 964 | SC 186        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.0s
[2025-01-30 07:49:34.764 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-01-30 07:49:34.764 (UTC+0100)] info/client        Selected endpoint with EndpointUrl opc.tcp://[hidden]:21381/MatrikonOpcUaWrapper, SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-01-30 07:49:34.764 (UTC+0100)] info/channel       TCP 964 | SC 186        | SecureChannel closed
[2025-01-30 07:49:34.780 (UTC+0100)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2025-01-30 07:49:34.780 (UTC+0100)] info/network       TCP 964 | Socket closed
[2025-01-30 07:49:34.889 (UTC+0100)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-01-30 07:49:34.889 (UTC+0100)] info/network       TCP 964 | Opening a connection to "10.104.174.90" on port 21381
[2025-01-30 07:49:34.889 (UTC+0100)] info/channel       TCP 964 | SC 187        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.0s
[2025-01-30 07:49:34.889 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-01-30 07:49:34.889 (UTC+0100)] info/client        Use the EndpointURL opc.tcp://[hidden]:21381/MatrikonOpcUaWrapper returned from FindServers and reconnect
[2025-01-30 07:49:34.889 (UTC+0100)] info/channel       TCP 964 | SC 187        | SecureChannel closed
[2025-01-30 07:49:34.889 (UTC+0100)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-01-30 07:49:34.889 (UTC+0100)] info/network       TCP 948 | Opening a connection to "10.104.174.90" on port 21381
[2025-01-30 07:49:34.905 (UTC+0100)] info/network       TCP 964 | Socket closed
[2025-01-30 07:49:34.905 (UTC+0100)] info/channel       TCP 948 | SC 188        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.0s
[2025-01-30 07:49:34.905 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-01-30 07:49:34.905 (UTC+0100)] info/client        Rejecting Endpoint 0: No matching UserTokenPolicy
[2025-01-30 07:49:34.905 (UTC+0100)] error/client       No suitable endpoint found
[2025-01-30 07:49:34.905 (UTC+0100)] info/client        Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
[2025-01-30 07:49:34.905 (UTC+0100)] info/channel       TCP 948 | SC 188        | SecureChannel closed
[2025-01-30 07:49:34.905 (UTC+0100)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
[2025-01-30 07:49:34.905 (UTC+0100)] info/network       TCP 948 | Socket closed

NoelGraf, 2025-02-14:
It would be helpful if you could provide some minimal code to see how you have configured the client.

RSpeets, 2025-02-15:
Here you will find the code that connects to the server:

```
bool OPCUAInitialize(const char* url, const int securityMode, const char* userName, const char* passWord, void* _ref, void** instance)
{
   *instance = NULL;
   UA_Client* client = UA_Client_new();
   References[client] = _ref;

   UA_ClientConfig_setDefault(UA_Client_getConfig(client));
   WriteProgress(Format("OPCUA connecting to %s", url), client);

   UA_StatusCode retval;

   UA_ClientConfig* cc = UA_Client_getConfig(client);

   if (ConsoleOutputEnabled) DebugEndPoints(client, url);

   cc->securityMode = (UA_MessageSecurityMode)securityMode;

   WriteProgress("Connect", client);
   retval = UA_Client_connect(client, url);

   if (retval != UA_STATUSCODE_GOOD) {
      string c = UA_StatusCode_name(retval);
      WriteProgress("Connection failed: " + c, client);

      UA_Client_delete(client);
      return false;
   }
   WriteProgress("Connection succesfull", client);
   *instance = client;
   return true;
}
```

RSpeets, 2025-03-13:
The code above was part of a DLL. I now isolated the issue within a few lines of code:

```
UA_Client *client;

int RunOPCUA() {
   UA_Client *client = UA_Client_new();
   UA_ClientConfig_setDefault(UA_Client_getConfig(client));

   /* Connect to a server */
   UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://ijmdv1srvopc:21381/MatrikonOpcUaWrapper");

   if (retval != UA_STATUSCODE_GOOD) {
      UA_Client_delete(client);
      return EXIT_FAILURE;
   }

   Sleep(1000);
   
   UA_Client_disconnect(client);
   UA_Client_delete(client);
   return EXIT_SUCCESS;
}
```

This returns the following output:

```
[2025-03-13 11:32:52.998 (UTC+0100)] warn/client        Skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-03-13 11:32:53.014 (UTC+0100)] info/network       TCP 420 | Opening a connection to "ijmdv1srvopc" on port 21381
[2025-03-13 11:32:53.029 (UTC+0100)] info/channel       TCP 420 | SC 80 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.0s
[2025-03-13 11:32:53.029 (UTC+0100)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-03-13 11:32:53.029 (UTC+0100)] info/client        The initially defined EndpointURL opc.tcp://ijmdv1srvopc:21381/MatrikonOpcUaWrapper is valid for the server
[2025-03-13 11:32:53.029 (UTC+0100)] info/client        Rejecting Endpoint 0: No matching UserTokenPolicy
[2025-03-13 11:32:53.029 (UTC+0100)] error/client       No suitable endpoint found
[2025-03-13 11:32:53.029 (UTC+0100)] info/client        Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
[2025-03-13 11:32:53.029 (UTC+0100)] info/channel       TCP 420 | SC 80 | SecureChannel closed
[2025-03-13 11:32:53.029 (UTC+0100)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadIdentityTokenRejected
[2025-03-13 11:32:53.029 (UTC+0100)] info/network       TCP 420 | Socket closed
```

As I indicated above: earlier versions of open62541 did connect ok.