open62541/open62541 #7063: Zephyr Example: Issues with Subscriptions and MonitoredItems - BadWaitingForInitialData

sudojh, 2025-01-30:
## Description
Hi,
I am building and running the zephyr server example from the master branch (commit 2fd81d6). My hardware is a nucleo_h563zi.
In the example, I changed the prj.conf to supporting Client/Server Subscriptions:
CONFIG_OPEN62541_ENABLE_SUBSCRIPTIONS=y

However, subscribing / monitoring the variable in UAExpert leads to a StatusCode "BadWaitingForInitialData" (Figure 1).
Simply browsing the variable and sending a ReadRequest although succeeds (Figure 2).

[image]
Figure 1

[image]
Figure 2

## Background Information / Reproduction Steps

Used CMakeLists.txt

````
cmake_minimum_required(VERSION 3.20.0)

find_package(Python REQUIRED COMPONENTS Interpreter)

set(DTC_OVERLAY_FILE ${CMAKE_CURRENT_LIST_DIR}/boards/nucleo_h563zi.overlay)

set(CONF_FILE ${CMAKE_CURRENT_LIST_DIR}/prj.conf)
set(EXTRA_ZEPHYR_MODULES ${CMAKE_CURRENT_SOURCE_DIR}/../../../arch)
find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})

project(open62541_server LANGUAGES C)

target_sources(app PRIVATE src/main.c)
target_link_libraries(app PUBLIC open62541)
````

Used prj.conf:
````
# Networking
CONFIG_NETWORKING=y
CONFIG_NET_DRIVERS=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_LOG=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=y
CONFIG_NET_TCP=y

CONFIG_NET_PKT_RX_COUNT=8
CONFIG_NET_PKT_TX_COUNT=8
CONFIG_NET_MAX_CONTEXTS=16
CONFIG_NET_MAX_CONN=16
CONFIG_NET_CONTEXT_SYNC_RECV=y
CONFIG_NET_TX_STACK_SIZE=4096
CONFIG_NET_RX_STACK_SIZE=4096
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_SOCKETS_POLL_MAX=12
CONFIG_NET_BUF_RX_COUNT=16
CONFIG_NET_BUF_TX_COUNT=16
CONFIG_NET_BUF_DATA_SIZE=128
CONFIG_NET_TC_TX_COUNT=1
CONFIG_NET_MAX_CONTEXTS=16
CONFIG_NET_MAX_CONN=16
CONFIG_NET_CONTEXT_SYNC_RECV=y
CONFIG_ENTROPY_GENERATOR=y

# Memory
CONFIG_HEAP_MEM_POOL_SIZE=200000
CONFIG_COMMON_LIBC_MALLOC_ARENA_SIZE=100000

CONFIG_INIT_STACKS=y
CONFIG_MAIN_STACK_SIZE=8192

# Debugging
CONFIG_DEBUG=y
CONFIG_STACK_SENTINEL=y

# Logging
CONFIG_LOG=y
CONFIG_PRINTK=y
CONFIG_COMMON_LIBC_MALLOC=y
CONFIG_NEWLIB_LIBC=y
CONFIG_POSIX_MAX_FDS=32
CONFIG_EVENTFD=y
CONFIG_EVENTFD_MAX=8
CONFIG_POSIX_API=y
CONFIG_SYS_HEAP_RUNTIME_STATS=y

CONFIG_OPEN62541=y
CONFIG_OPEN62541_ENABLE_DISCOVERY_SEMAPHORE=n
CONFIG_OPEN62541_MULTITHREADING=0
CONFIG_OPEN62541_NAMESPACE_ZERO="MINIMAL"
CONFIG_OPEN62541_ENABLE_PUBSUB=n
CONFIG_OPEN62541_ENABLE_PUBSUB_INFORMATIONMODEL=n
CONFIG_OPEN62541_ENABLE_SUBSCRIPTIONS_EVENTS=n
CONFIG_OPEN62541_ENABLE_SUBSCRIPTIONS=y
CONFIG_OPEN62541_ENABLE_HISTORIZING=n
CONFIG_OPEN62541_ENABLE_JSON_ENCODING=n
````

Log Output

```` 
*** Booting Zephyr OS build v3.6.0 ***
[1970-01-01 00:00:02.002 (UTC+0000)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[1970-01-01 00:00:02.012 (UTC+0000)] info/server        AccessControl: Anonymous login is enabled
[1970-01-01 00:00:02.020 (UTC+0000)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[1970-01-01 00:00:02.045 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;s=the.answer): No TypeDefinition. Use the default TypeDefinition for the Variable/Object
[1970-01-01 00:00:02.062 (UTC+0000)] info/network       TCP     | Listening on all interfaces
[1970-01-01 00:00:02.069 (UTC+0000)] info/network       TCP 4   | Creating listen socket for "0.0.0.0" on port 4840
[1970-01-01 00:00:02.079 (UTC+0000)] info/server        New DiscoveryUrl added: opc.tcp://0.0.0.0:4840
[00:00:02.001,000] <inf> open62541_test: Starting UA_Server
[1970-01-01 00:00:05.396 (UTC+0000)] info/network       TCP 5   | Connection opened from "192.168.1.149" via the server socket 4
[1970-01-01 00:00:05.407 (UTC+0000)] info/channel       TCP 5   | SC 1  | SecureChannel created
[1970-01-01 00:00:05.416 (UTC+0000)] info/channel       TCP 5   | SC 1  | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.0s
[1970-01-01 00:00:05.434 (UTC+0000)] info/channel       TCP 0   | SC 0  | SecureChannel closed
[1970-01-01 00:00:05.442 (UTC+0000)] info/network       TCP 5   | Socket closed
[1970-01-01 00:00:06.133 (UTC+0000)] info/network       TCP 5   | Connection opened from "192.168.1.149" via the server socket 4
[1970-01-01 00:00:06.144 (UTC+0000)] info/channel       TCP 5   | SC 2  | SecureChannel created
[1970-01-01 00:00:06.153 (UTC+0000)] info/channel       TCP 5   | SC 2  | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.0s
[1970-01-01 00:00:06.172 (UTC+0000)] info/channel       TCP 0   | SC 0  | SecureChannel closed
[1970-01-01 00:00:06.179 (UTC+0000)] info/network       TCP 5   | Socket closed
[1970-01-01 00:00:09.039 (UTC+0000)] info/network       TCP 5   | Connection opened from "192.168.1.149" via the server socket 4
[1970-01-01 00:00:09.050 (UTC+0000)] info/channel       TCP 5   | SC 3  | SecureChannel created
[1970-01-01 00:00:09.059 (UTC+0000)] info/channel       TCP 5   | SC 3  | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.0s
[1970-01-01 00:00:09.078 (UTC+0000)] info/channel       TCP 0   | SC 0  | SecureChannel closed
[1970-01-01 00:00:09.085 (UTC+0000)] info/network       TCP 5   | Socket closed
[1970-01-01 00:00:09.099 (UTC+0000)] info/network       TCP 5   | Connection opened from "192.168.1.149" via the server socket 4
[1970-01-01 00:00:09.110 (UTC+0000)] info/channel       TCP 5   | SC 4  | SecureChannel created
[1970-01-01 00:00:09.119 (UTC+0000)] info/channel       TCP 5   | SC 4  | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.0s
[1970-01-01 00:00:09.137 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;g=5704b11b-e7d0-b3b5-d72d-a20a57355305): Parent node not found
[1970-01-01 00:00:09.152 (UTC+0000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=1;g=5704b11b-e7d0-b3b5-d72d-a20a57355305): The parent reference for is invalid
[1970-01-01 00:00:09.168 (UTC+0000)] warn/session       TCP 5   | SC 4  | Session "urn:jho:UnifiedAutomation:UaExpert"  | Creating the session diagnostics object failed with StatusCode BadParentNodeIdInvalid
[1970-01-01 00:00:09.185 (UTC+0000)] info/session       TCP 5   | SC 4  | Session "urn:jho:UnifiedAutomation:UaExpert"  | Session created
[1970-01-01 00:00:09.197 (UTC+0000)] info/session       TCP 5   | SC 4  | Session "urn:jho:UnifiedAutomation:UaExpert"  | ActivateSession: Session activated with ClientUserId ""
[1970-01-01 00:00:11.950 (UTC+0000)] info/session       TCP 5   | SC 4  | Session "urn:jho:UnifiedAutomation:UaExpert"  | Subscription 1 | Subscription created (Publishing interval 500.0ms, max 1000 notifications per publish)
[1970-01-01 00:00:12.261 (UTC+0000)] info/session       TCP 5   | SC 4  | Session "urn:jho:UnifiedAutomation:UaExpert"  | Subscription 1 | MonitoredItem 1 | Created the MonitoredItem (Sampling Interval: 250.0ms, Queue Size: 1)
````

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [x ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2025-02-05:
Hmm. I have never seen a problem with the sample callbacks.
Are you sure your system time is advancing like it should?

sudojh, 2025-02-05:
Hi,

I changed:
UA_DateTime interval = (UA_DateTime)(interval_ms * UA_DATETIME_MSEC);
 ... to ...
UA_DateTime interval = (UA_DateTime)interval_ms;
in open62541/arch/common/timer.c

While receiving "now" in ms since system restart, the interval was in ms*10000. Therefore the callback time was "never" reached.

closed 2025-02-05 by sudojh