open62541/open62541 #7065: Unable to connect to OPCUA server with encryption enabled

Shobhit-exalens, 2025-01-30:
Hello,
I've been developing an OPC UA client to discover and connect to various OPC UA server endpoints. While I successfully connected to a server with None security mode and encryption, I’m struggling to connect to the Ignition OPC UA server on localhost:62541. The same issue occurs when attempting to discover its endpoints.
Discovery:

[image]

Connection:

[image]

**Library Build** (version: v1.4.8)
1. cmake -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_ENCRYPTION=OPENSSL ..
2. make open62541-amalgamation-source 
3. make open62541-amalgamation-header

### client_connect.c

[image]

I also tried using the Open62541 example script (client_encryption.c), but the result was the same. However, when I compiled same custom discoverEndpoint.c with the single-file release (open62541.c and open62541.h), I was able to successfully discover the Ignition server’s endpoints.

[image]

Any help would be greatly appreciated!

NoelGraf, 2025-02-27:
If you are connecting to an existing server, it is of course important how the server is configured. Without logging messages, it is difficult to tell what exactly is not working.
Can you provide more information?

shobhit-datumspace, 2025-02-28:
[image]
[image]

@NoelGraf  ,this is my current ignition server configuration with security policy "Basic256Sha256" but when i changed it to "None,Basic256Sha256" i was able to connect to the server after moving the Quarantined Certificates to Trusted Certificates. 
I want to connect to the server withouf changing the security policy to None.

NoelGraf, 2025-03-07:
Unfortunately, I can't see anything specific in the server logs provided. The problem is that the response contains the very general StatusCode Bad_SecurityChecksFailed, which doesn't give much insight into the root cause.

If you enable the None security policy and connect to the server successfully, are you actually connecting through the Basic256Sha256 endpoint?

shobhit-datumspace, 2025-04-02:
@NoelGraf Yes, i was connecting through Basic256sha256 endpoint.
I am able to connect to same ignition server using a python client and also using UA-expert without adding "None" Security policy.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier