open62541/open62541 #7067: OPC DA client support

oktonion, 2025-01-31:
Hello, I have been trying to understand all differences between this implementation of server (UA) and classic (DA). Despite all docs and info online it doesn't quite highlight differences in protocol client access:
1. Can I use old OPC DA client to connect to this UA server? Why not/what are the differences?
2. If not are there any c/c++ or other converters of OPC DA client requests for OPC UA server?(f.e. some opc_da_to_opc_ua_client_wrapper or opc_da_to_opc_ua_server_wrapper)
3. As for codebase itself - could I add OPC DA server interface somehow to existing open62541 server?

Thanks in advance for thoughtful reply to my maybe little naive questions.

RolfKal, 2025-01-31:
OPC DA is a completely different architecture to OPC UA. The low level protocol for OPC DA is based on Windows Object Linking and Embedding (OLE) as transport layer whereas OPC UA is usually directly based on a TCP/IP or UDP transport layer.

So to your questions:

1) No
2) Maybe, but I would think it's a very complicated and cumbersome project to do. Most likely nobody felt like doing that.
3) No, that is simply not feasable.

oktonion, 2025-02-01:
So there is nothing like [this](https://opcexpert.com/support/convert-opc-data) as open source implementation? 
I understand that architecture is completely different, however it seems odd that despite same name "OPC" classic DA and UA protocols for client connections to server for high lever read/write data operations to low level controllers is **completely** different. Misguiding as for me.

V-Strassheim, 2025-02-03:
I've never used it, but maybe this would suit your application: https://rapidscada.org/product/connectivity-and-integration/

o6-nbeier, 2026-10-20:
Closing because of not related.

closed 2026-10-20 by o6-nbeier