open62541/open62541 #7073: File ua_config_json.c  , function parseJSONConfig() : no check of cj5_parse() result.

Berendej, 2025-02-03:
## Description
File ua_config_json.c  , function parseJSONConfig() : no check of cj5_parse() result.
It would be nice to check it like:

```
static UA_StatusCode
parseJSONConfig(UA_ServerConfig *config, UA_ByteString json_config) {
    ...
    cj5_result r = cj5_parse(json, (unsigned int)json_config.length, tokens, MAX_TOKENS, NULL);
    /* vvvvv something like this vvvv */
    if (CJ5_ERROR_NONE != r.error) {
        return UA_STATUSCODE_BAD;
    }
   /* ^^^^^^^^^^^^^^^^^^^^^^^^^ */
```
otherwise if parsing fails we still get UA_STATUSCODE_GOOD since no tokens are present.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.4.8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2025-02-05:
Sounds good.
Are you in a position to open a PR for that?
Otherwise we can handle it.

Berendej, 2025-02-06:
I think that the modification is quite simple and will require less effort if it will be done by maintainers.