open62541/open62541 #7082: nodeset file containing UAVariable without value is loaded into server and read from client always reads NULL Value.

Graos, 2025-02-04:
**OPEN62541 OPC UA STACK VERSION : v1.4.8-6-g0cc73feb0** 

When a nodeset file with two array variables (DateTime and Double) one with initialized value (Double type)  and the other( DateTime) without any value is loaded and the same is read from an OPC UA Client the signal without a value is null but its attributes are intact and valid.

   <UAVariable UserAccessLevel="3" DataType="DateTime" BrowseName="1:PV_DataTime_Array" ArrayDimensions="5" ValueRank="1" NodeId="ns=1;i=6017" AccessLevel="3">
        <DisplayName>PV_DataTime_Array</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=5003</Reference>
        </References>
    </UAVariable>
    <UAVariable UserAccessLevel="3" DataType="Double" BrowseName="1:PV_Double_Array" ArrayDimensions="5" ValueRank="1" NodeId="ns=1;i=6019" AccessLevel="3">
        <DisplayName>PV_Double_Array</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference IsForward="false" ReferenceType="HasComponent">ns=1;i=5003</Reference>
        </References>
        <Value>
            <uax:ListOfDouble>
                <uax:Double>99.99</uax:Double>
                <uax:Double>99.99</uax:Double>
                <uax:Double>99.99</uax:Double>
                <uax:Double>99.99</uax:Double>
                <uax:Double>99.99</uax:Double>
            </uax:ListOfDouble>
        </Value>
    </UAVariable>

[image]

when DateTime is read from an OPC UA Client(UaExpert) here

[image]

[image]

jpfr, 2025-02-05:
Right.
For the scalar NodeRank we generate a "default" scalar.
But it appears we don't for arrays.

We can generate a default array for the ArrayDimensions you have.
Is this what you meant?

Graos, 2025-02-05:
generating a default array value for the array variable without a value in the loaded nodeset file will be great. For your information even for the scalar rank, variable of type statusCode which does not have a value results in null when read from client.

[image]