open62541/open62541 #7094: nodeset: Adding UAFX data namespace fails

V-Strassheim, 2025-02-05:
## Description
When adding [UAFX nodeset](https://github.com/OPCFoundation/UA-Nodeset/tree/9cd8699c3addcdb01f5a45a7c7352453ee6838ad/UAFX) to the nodesetloader example, the following error message appears at runtime:
```
info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=3;i=5019): Parent node not found
info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=3;i=5019): The parent reference for is invalid
info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=3;i=7032): Parent node not found
info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode (ns=3;i=7032): The parent reference for is invalid
... 
Follow-up errors for further nodes
...
error/server       Adding the UAFX Data namespace failed. Please check previous error output.
```

For ns=3;i5019 the parent node i=15606 is obviously missing, but this should come along with the NS0 (FULL) which is defined in /deps/ua-nodeset/Schema/[Opc.Ua.Nodest2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/9cd8699c3addcdb01f5a45a7c7352453ee6838ad/Schema/Opc.Ua.NodeSet2.xml).

Does anyone have any idea what else I might be missing?

## Background Information / Reproduction Steps

This is my CMakeLists.txt (I changed it in examples/nodeset)
```CMake
####################
# Nodeset Examples #
####################

###################
# UAFX Nodeset #

set(FILE_CSV_DIRPREFIX ${PROJECT_SOURCE_DIR}/nodeset)
set(FILE_BSD_DIRPREFIX ${PROJECT_SOURCE_DIR}/nodeset)
set(FILE_NS_DIRPREFIX ${PROJECT_SOURCE_DIR}/nodeset)

if(NOT CMAKE_PROJECT_NAME STREQUAL PROJECT_NAME)
    # needed or cmake doesn't recognize dependencies of generated files
    get_target_property(OPEN62541_BIN_DIR open62541::open62541 BINARY_DIR)
    set(PROJECT_BINARY_DIR ${OPEN62541_BIN_DIR})
endif()
###################

if(NOT CMAKE_PROJECT_NAME STREQUAL PROJECT_NAME)
    set(FILE_CSV_DIRPREFIX ${UA_NODESET_DIR})
    set(FILE_BSD_PLCOPEN_DIRPREFIX ${UA_NODESET_DIR})
    set(FILE_BSD_POWERLINK_DIRPREFIX ${PROJECT_SOURCE_DIR}/nodeset)
    set(FILE_NS_DIRPREFIX ${UA_NODESET_DIR})
    set(FILE_UAFX_DIRPREFIX ${UA_NODESET_DIR})
else()
    set(FILE_CSV_DIRPREFIX ${UA_NODESET_DIR})
    set(FILE_BSD_PLCOPEN_DIRPREFIX ${UA_NODESET_DIR})
    set(FILE_BSD_POWERLINK_DIRPREFIX ${PROJECT_SOURCE_DIR}/nodeset)
    set(FILE_NS_DIRPREFIX ${UA_NODESET_DIR})
    set(FILE_UAFX_DIRPREFIX ${UA_NODESET_DIR})
endif()

# PLCopen requires the full ns0 as basis
if(UA_NAMESPACE_ZERO STREQUAL "FULL")

    # Generate types and namespace for DI
    ua_generate_nodeset_and_datatypes(
        NAME "di"
        FILE_CSV "${FILE_CSV_DIRPREFIX}/DI/Opc.Ua.Di.NodeIds.csv"
        FILE_BSD "${FILE_BSD_PLCOPEN_DIRPREFIX}/DI/Opc.Ua.Di.Types.bsd"
        FILE_NS "${FILE_NS_DIRPREFIX}/DI/Opc.Ua.Di.NodeSet2.xml"
        INTERNAL
    )

    ua_generate_nodeset_and_datatypes(
        NAME "fxdata"
        FILE_CSV "${FILE_CSV_DIRPREFIX}/UAFX/opc.ua.fx.data.nodeset2.csv"
        FILE_BSD "${FILE_UAFX_DIRPREFIX}/UAFX/opc.ua.fx.data.nodeset2.bsd"
        FILE_NS "${FILE_NS_DIRPREFIX}/UAFX/opc.ua.fx.data.nodeset2.xml"
        IMPORT_BSD "UA_TYPES#${CMAKE_CURRENT_LIST_DIR}/../../deps/ua-nodeset/Schema/Opc.Ua.Types.bsd"
        DEPENDS "di"
        INTERNAL
    )

    ua_generate_nodeset_and_datatypes(
        NAME "fxcm"
        FILE_CSV "${FILE_CSV_DIRPREFIX}/UAFX/opc.ua.fx.cm.nodeset2.csv"
        FILE_BSD "${FILE_UAFX_DIRPREFIX}/UAFX/opc.ua.fx.cm.nodeset2.bsd"
        FILE_NS "${FILE_NS_DIRPREFIX}/UAFX/opc.ua.fx.cm.nodeset2.xml"
        IMPORT_BSD "UA_TYPES#${CMAKE_CURRENT_LIST_DIR}/../../deps/ua-nodeset/Schema/Opc.Ua.Types.bsd"
        DEPENDS "di"
        DEPENDS "fxdata"
        INTERNAL
    )

    ua_generate_nodeset_and_datatypes(
        NAME "fxac"
        FILE_CSV "${FILE_CSV_DIRPREFIX}/UAFX/opc.ua.fx.ac.nodeset2.csv"
        FILE_BSD "${FILE_UAFX_DIRPREFIX}/UAFX/opc.ua.fx.ac.nodeset2.bsd"
        FILE_NS "${FILE_NS_DIRPREFIX}/UAFX/opc.ua.fx.ac.nodeset2.xml"
        IMPORT_BSD "UA_TYPES#${CMAKE_CURRENT_LIST_DIR}/../../deps/ua-nodeset/Schema/Opc.Ua.Types.bsd"
        DEPENDS "di"
        DEPENDS "fxcm"
        INTERNAL
    )

    add_example(server_nodeset_uafx server_nodeset_uafx.c
                ${UA_NODESET_DI_SOURCES}
                ${UA_NODESET_FXCM_SOURCES}
                ${UA_NODESET_FXDATA_SOURCES}
                ${UA_NODESET_FXAC_SOURCES}
                ${UA_TYPES_DI_SOURCES}
                ${UA_TYPES_FXCM_SOURCES}
                ${UA_TYPES_FXDATA_SOURCES}
                ${UA_TYPES_FXAC_SOURCES}
                )
    add_dependencies(server_nodeset_uafx open62541-generator-ns-di open62541-generator-ns-fxcm open62541-generator-ns-fxdata open62541-generator-ns-fxac)
endif()

if(UA_ENABLE_NODESET_INJECTOR)
    set(UA_NODESETINJECTOR_GENERATORS ${UA_NODESETINJECTOR_GENERATORS} PARENT_SCOPE)
    set(UA_NODESETINJECTOR_SOURCE_FILES ${UA_NODESETINJECTOR_SOURCE_FILES} PARENT_SCOPE)
    set(UA_NODESETINJECTOR_EXAMPLE_NAMES ${UA_NODESETINJECTOR_EXAMPLE_NAMES} PARENT_SCOPE)
endif()

```

And here is the C-File:

```C
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include "open62541/namespace_fxcm_generated.h"
#include "open62541/namespace_di_generated.h"
#include "open62541/namespace_fxdata_generated.h"
#include "open62541/namespace_fxac_generated.h"
#include "open62541/namespace0_generated.h"

#include <signal.h>
#include <limits.h>
#include <stdlib.h>

UA_Boolean running = true;

static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(int argc, char** argv) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    /* create nodes from nodeset */
    size_t idx = LONG_MAX;
    UA_StatusCode retval = 0;
    retval = UA_Server_getNamespaceByName(server, UA_STRING("http://opcfoundation.org/UA/DI/"), &idx);
    if(retval != UA_STATUSCODE_GOOD) {
        retval = namespace_di_generated(server);
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                         "Adding the DI namespace failed. Please check previous error output.");
            UA_Server_delete(server);
            return EXIT_FAILURE;
        }
    }

    retval = UA_Server_getNamespaceByName(server, UA_STRING("http://opcfoundation.org/UA/FX/Data/"), &idx);
    if(retval != UA_STATUSCODE_GOOD) {
        retval = namespace_fxdata_generated(server);
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                         "Adding the UAFX Data namespace failed. Please check previous error output.");
            UA_Server_delete(server);
            return EXIT_FAILURE;
        }
    }

    retval = UA_Server_getNamespaceByName(server, UA_STRING("http://opcfoundation.org/UA/FX/CM/"), &idx);
    if(retval != UA_STATUSCODE_GOOD) {
        retval = namespace_fxcm_generated(server);
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                         "Adding the UAFX CM namespace failed. Please check previous error output.");
            UA_Server_delete(server);
            return EXIT_FAILURE;
        }
    }

    retval = UA_Server_getNamespaceByName(server, UA_STRING("http://opcfoundation.org/UA/FX/AC/"), &idx);
    if(retval != UA_STATUSCODE_GOOD) {
        retval = namespace_fxac_generated(server);
        if(retval != UA_STATUSCODE_GOOD) {
            UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                         "Adding the UAFX AC namespace failed. Please check previous error output.");
            UA_Server_delete(server);
            return EXIT_FAILURE;
        }
    }

    retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

Used CMake options:

```bash
cmake -DUA_ARCHITECTURE=posix -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON -DUA_ENABLE_NODESETLOADER=ON -DUA_ENABLE_DA=ON -DUA_ENABLE_DATATYPES_ALL=ON -DUA_LOGLEVEL=100 ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Operating system: Debian
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Self-contained code example attached

V-Strassheim, 2025-02-06:
i=15606 is RoleSet node, which is not implemented yet.
Here is the reference in code:
- https://github.com/open62541/open62541/blob/43178b68147d211f117e358612d28efaa39b072d/src/server/ua_server_ns0.c#L1127
And some additional info:
- https://github.com/open62541/open62541/pull/6521
- https://groups.google.com/g/open62541/c/4z1iJAwoqwU

closed 2025-02-06 by V-Strassheim

tlf30, 2026-03-03:
I’m very interested in getting UAFX to work. Are there plans to merge the PR to support role based access control?