open62541/open62541 #7098: allowNonePolicyPassword = false doesn't remove username policy from SecurityPolicy#None endpoint

tomitkl, 2025-02-07:
## Description

I'm not exactly sure how allowNonePolicyPassword  should work but if I set it to false and add encrypted security policies and security policy none, I can see the endpoint with SecurityPolicy#None has open62541-username-policy-none#Basic256Sha256 even if I have tried to disable it with the flag:

[image]

If I invert the getDefaultEncryptedSecurityPolicy check in updateEndpointUserIdentityToken added in #6837, it seems to work so that the username policy is removed:

[image]

So, is the check correct at the moment or should the username policy be removed from the SecurityPolicy#None if there are secure alternatives and only allowed if there are no secure ways to connect?

## Background Information / Reproduction Steps

This is about how I configure the security polices/endpoints/access control for the server:

```
UA_ServerConfig_addSecurityPolicies_Filestore(&config, &certificate, &key, storePath);
config.allowNonePolicyPassword = false;
UA_ServerConfig_addSecurityPolicyNone(&config, nullptr);
UA_ServerConfig_addAllEndpoints(&config);
UA_AccessControl_defaultWithLoginCallback(&config, true, &config.securityPolicies[config.securityPoliciesSize - 1].policyUri, 1, login, &loginCallback, this);
```

This is the change I did to change the behavior:

```
--- a/open62541/src/server/ua_services_discovery.c
+++ b/open62541/src/server/ua_services_discovery.c
@@ -351,8 +351,8 @@ updateEndpointUserIdentityToken(UA_Server *server, UA_EndpointDescription *ed) {
            !(sc->allowNonePolicyPassword && utp->tokenType == UA_USERTOKENTYPE_USERNAME) &&
            UA_String_equal(&ed->securityPolicyUri, &UA_SECURITY_POLICY_NONE_URI)) {
             UA_SecurityPolicy *encSP = getDefaultEncryptedSecurityPolicy(server);
-            if(!encSP) {
-                /* No encrypted SecurityPolicy available */
+            if(encSP) {
+                /* Encrypted SecurityPolicy available */
                 UA_LOG_WARNING(sc->logging, UA_LOGCATEGORY_CLIENT,
                                "Removing a UserTokenPolicy that would allow the "
                                "password to be transmitted without encryption "
```

Used CMake options: At least UA_ENABLE_ENCRYPTION

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): current master and probably 1.4 branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: own embedded system
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2025-02-13:
The allowNonePolicyPassword setting does not remove the None endpoint, it blocks username/password authentication through that endpoint. This ensures that passwords are not transmitted in clear text. To completely remove the None policy, use the UA_ServerConfig_addAllSecureSecurityPolicies function, which adds only secure policies.

tomitkl, 2025-02-19:
OK, doesn't really matter for us now as we decided remove Security#None for all other than discovery i.e. use securityPolicyNoneDiscoveryOnly = true.

closed 2025-02-19 by tomitkl