open62541/open62541 #7102: Performance heavy timing problems in epoll handling

MarkusBeerwerth, 2025-02-11:
## Description

Hi,

after upgrading from a 1.3 to the newest 1.4 version on a Linux system I noticed some performance issues.

In my case the average cpu time of the server increased around 20 times compared to the 1.3 version, even noticeable when running a idle server without any connections. So i gave a look into it.

After some investigation trying different 1.4 versions and recording traces all 1.4 versions shows the same behavior, that seems to be caused by the switch from the ```select``` to ```epoll``` kernel apis.

On every ```epoll_wait``` wakeup the opc server does many additional noblocking epoll_wait calls with timeout 0. Way to many to count in my case it seems to be something in the high 2 digit to low hundreds range adding usually up to more 500us of cpu time in each server call. So the overhead usually adds to way more that the expected cpu load of the server itself.

After building the library and debugging it following issue seems to occurre in the ```eventloop_posix.c``` and ```eventloop_posix_epoll.c``` modules in the functions ```UA_EventLoopPOSIX_run``` ```UA_EventLoopPOSIX_pollFDs```:

- all the timeouts of the internally added callbacks are managed with a precision of 100 ns
- the used ```epoll_wait``` only supports a precision of 1ms. The input is rounded down by integer division
-  ```epoll_wait``` will wake up a ms too early, so the server main loop in ```UA_Server_run``` will loop nonblocking until the timeout a few hundred us away in the future is fulfilled, the callback executed and the timeout reset.

Depends a bit on the starting time and jitter, but worst case any callback takes around 1ms of extra time per execution indepedent of system performance, creating a noticeable amount of overhead. Even if ```epoll``` only supports ms precision it usually triggers rather precise with minimal jitter in the double digit microsecond range x ms after it got called. So how big the overhead gets, is consistent after the first call.

In all it should be a general problem that effects all linux builds in 1.4.x version.

## Background Information / Reproduction Steps
- run a server on Linux. 
- monitor cpu usage
- record traces with kernel events (trace-cmd, lttng, etc.)
- count timouts < 1ms in ```UA_EventLoopPOSIX_run```

Issue #6593 seems to be related 

Used CMake options (should be independent these):
- Default debug Build config
- tested on both arm64 and x86-64

## Possible Solution:
At first glance I see a few possible options to fix the problem, but maybe you have better ideas.

**Use epoll_pwait2:**
There is already a commented out `epoll_pwait2` implementation right under the `epoll_wait` implementation in ```UA_EventLoopPOSIX_pollFDs```
```epoll_pwait2``` would probably work fine, but could be a breaking change for older systems. Already encountered some older linux versions that didn't support it before.

**Ceil the timout that is going into epoll**
A rather quick and dirty solution would be to ceil the timeout that is going into ```epoll_wait```
tried it as a temporary workaround and seems to work rather good in my case. Subsequent calls of the repeated callbacks will usually have the same offset to the ```epoll_wait``` call so they will still get called rather precise in x ms intervals. It would also only effect the linux version.

**Manage the timouts with ms precision:**
A more clean solution would probably be to manage the callback times only with ms precision and take into account the jitter by checking if the timestamp is near the expected one. They only get parameterized with a ms precision. Not sure if there a good usecases to get more precise than that on different plattforms.

## User Workarounds:

Run `UA_Server_run_iterate` with the nonblocking `waitInternal`  yourself and sleep a ms longer than it return

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):  All  1.4.x Versions
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian Bookworm
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2025-02-21:
We can reproduce.
This will get fixed.

jpfr, 2025-03-06:
The fix is in #7150.
Please validate if that solves the issue for you.

MarkusBeerwerth, 2025-03-19:
Not sure if this really fixes the problem. May I'm missing something but batching the timers and increasing the intervals of the server calls looks more like a mitigation to me not fixing the underlying busy polling issue. Or is there an implicit way that I'm missing how the timers get scheduled to full milliseconds?

I exspected more something like that.
``` cpp
UA_StatusCode
UA_EventLoopPOSIX_pollFDs(UA_EventLoopPOSIX *el, UA_DateTime listenTimeout) {
    UA_assert(listenTimeout >= 0);

    //round up to prevent busy polling
    bool roundup = listenTimeout % UA_DATETIME_MSEC != 0;
    int timeoutMs = (int)(listenTimeout / UA_DATETIME_MSEC);
    timeoutMs = roundup ? timeoutMs + 1: timeoutMs;

    /* Poll the registered sockets */
    struct epoll_event epoll_events[64];
    int epollfd = el->epollfd;
    UA_UNLOCK(&el->elMutex);
    int events = epoll_wait(epollfd, epoll_events, 64,
        timeoutMs);
    //...
```

Maybe my description also wasn`t that clear and we have a little missunderstanding. What I see within traces is something like that:

[image]

So in that example every server call wastes around 850us cpu time execution for a callback that takes tens of microseconds. That was on a raspberry py on a full fledged x86 system the ratios would probably be even worse.

After the sugested fix a similar servercall won`t do the busy polling you would get this picture:

[image]

In that specific case it took 40us so you would use up 20 times less cpu time. Can`t really tell if it is the same type of callback but from what I saw they usually take less than 100us.

The difference would be that it wakes up after the scheduled callback time. It's is within the last millisecond, so you would loose some precision over when they get called

jpfr, 2025-03-19:
This was addressed in https://github.com/open62541/open62541/pull/7150.
This has already landed on the 1.4 branch.

The main problem was that we did not "batch" the callbacks with identical intervals and nearly the same base timestamp.

Does the change from the PR solve your issue?

MarkusBeerwerth, 2025-03-19:
Ok just tested it and would say did a bit more than I exspected. Lowering the CPU times to acceptable levels.

But like I mean, the underlying issue isn't really fixed. There is still a phase of busy polling before each callback
[image]

Due too the batched nature the wakeups now doesn't occure that often and the overhead isn't rising with the number of callbacks. So 1 subscription has the same undelying overhead than 1000 subscriptions with the same interval.

Fixing the busy poll still would reduce the cpu time by some big margin in most cases.

But now it already should be in the range I could live with

jpfr, 2025-03-19:
Right.
I add some dumb printf logging and did not see the same behavior on my box.
Just a single sleep between any timed callbacks.
But that be due to developing on a virtual machine where the underlying timer interrupts might be rate limited.

My suggestion is that we make a change (only) to the POSIX subsystem, such that we grab+execute also all future timed callbacks within a 1ms window.

That should get rid of the busy loop.

jpfr, 2025-03-19:
Reopening the issue, as this should be addressed.

jpfr, 2025-03-22:
@MarkusBeerwerth Please check PR #7180.
This should get fully rid of the busy loop.

commit 041d6a21 (2025-03-25): fix(core): Fix a busy-loop when timeouts are less than 1ms (fixes #7102)
closed 2025-03-25 by jpfr