open62541/open62541 #7123: Data source change notification

xR3b0rn, 2025-02-20:
If I write to a variable, all monitored items are notified immediately. However, when using data source callbacks, I cannot trigger the same immediate notification to the server as is done with the regular node store.

Shouldn't be there a dedicated function that I can call once my variable changes?

undroidwish, 2025-02-22:
I second this feature request. For my Tcl binding "topcua" which is still at open62541 version 1.3.x I patched this
[snippet](https://www.androwish.org/home/file?ci=tip&name=jni/topcua/open62541/chw.patch&ln=318-335) for similar reasons. Maybe the development team would like to pick up this idea.

xR3b0rn, 2025-02-22:
I currently use DataSource callbacks that only move—or perform a shallow copy of—the UA_Variant. The caller of the UA_Server_writeValue function is the true owner of the memory referenced by the UA_Variant’s data pointer.

I struggled a lot to figure this out, but this approach results in zero copying. I still believe that my original idea of incorporating a notify function would be a valuable addition.