open62541/open62541 #7131: Integration Testing Without Real-Time Sleep

kmugEq, 2025-02-24:
Hi,

In my integration tests, I want to verify scenarios such as connection, disconnection, and loss of connection.  I do the same many of the existing tests in open62541 by setting up a server and client. But I can not use UA_FakeSleep etc.  Is there an elegant way to achieve this without relying on real-time sleep?

I would prefer to avoid building a separate version of open62541 for integration tests versus production.

Are there any recommended approaches or best practices for handling these cases efficiently?

Thanks!

jpfr, 2025-03-02:
The fake sleep allows us to have unit tests for timeout scenarios without having to actually wait for minutes and hours.

Are there scenarios within open62541 that you cannot recreate or is this about the interaction with another private system that you have?

kmugEq, 2025-03-13:
Yes this is the interaction with my system.
Testing that the client, during disconnects, server restarts etc. always will reconnect and continue.
Especially subscriptions. 

Thanks!

o6-nbeier, 2026-10-20:
Thank you for the question.

On current branches you can configure the EventLoop's clock and advance it in your test instead of sleeping, while older branches need a different clock substitution (`UA_fakeSleep` is internal test code).

Closing for now.

closed 2026-10-20 by o6-nbeier