open62541/open62541 #7139: libc_time.c __secs_to_tm overrides function of the same name in musl

jwalt, 2025-02-27:
## Description

Using open62541 with musl libc breaks `gmtime` and similar functions in musl, because `libc_time.c` was taken straight from musl and kept the function name. However, the function is from a very old musl version and is incompatible with current versions of musl. As it takes precedence during linking, it breaks date-related functions in musl.

Given that names starting with an underscore are reserved identifiers, a simple rename of `__secs_to_tm` could be a proper fix.

## Background Information / Reproduction Steps

see above

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux with musl libc
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2025-03-02:
Understood.

Can you make a PR which does the renaming and updates to a more recent upstream from musl?

jpfr, 2025-04-01:
Solved via #7185.

commit df61c22f (2025-04-01): fix(deps): Update musl time methods. Prefix to avoid name clash (fix #7139)
closed 2025-04-01 by jpfr