open62541/open62541 #7141: docker container build scripts

jackybek, 2025-03-02:
I created a docker container that automatically
1. pulls the OS image e.g. debian:bookworm
2. updates the OS to the latest patch
3. installs the necessary libraries for compilation
4. updates openssl to the latest version
5. installs websocket libraries
6. install other libraries that augment open62541 options e.g. mqtt, json, xml
7. pull the latest open62541 from github and builds the base library
8. run cmake to turn on open62541 options and builds it

To test, i also added a simple program that builds a Local Directory Server using open62541.
Source files are pulled from github.

jpfr, 2025-03-02:
Sounds good.
You want to make that available?

jackybek, 2025-03-03:
Currently i faced these 3 errors during the CMake process using the dockerfile:

CMake Warning:
  Manually-specified variables were not used by the project:

    UA_ENABLE_DISCOVERY_MULTICAST_MDNSD
    UA_ENABLE_GDS_PUSHMANAGEMENT
    UA_ENABLE_JSON_ENCODING_LEGACY

Because of the first warning, when i compile my application code, it fails.

Other than that, the other scripts are running fine.
Once I solve the above CMake error, i should be able to complete the docker packaging process.

[steps to use debian container.txt](https://github.com/user-attachments/files/19047042/steps.to.use.debian.container.txt)

jackybek, 2025-03-05:
Hi Julius,
I am getting this funny error when launching the LDS from within the docker container.
[2025-03-05 00:20:33.555 (UTC+0000)] info/eventloop     Starting the EventLoop
[2025-03-05 00:20:33.557 (UTC+0000)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2025-03-05 00:20:33.557 (UTC+0000)] info/server        AccessControl: Anonymous login is enabled
[2025-03-05 00:20:33.557 (UTC+0000)] warn/server        x509 Certificate Authentication configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2025-03-05 00:20:33.566 (UTC+0000)] error/server       **Initialization of Namespace 0 (after bootstrapping) failed with BadNodeIdUnknown. See previous outputs for any error messages.**
[2025-03-05 00:20:33.566 (UTC+0000)] info/server        PubSub cleanup was called.
[2025-03-05 00:20:33.566 (UTC+0000)] info/eventloop     Stopping the EventLoop
[2025-03-05 00:20:33.566 (UTC+0000)] info/network       MQTT    | Shutting down the ConnectionManager
[2025-03-05 00:20:33.567 (UTC+0000)] info/network       ETH     | Shutting down the ConnectionManager
[2025-03-05 00:20:33.567 (UTC+0000)] info/network       UDP     | Shutting down the ConnectionManager
[2025-03-05 00:20:33.567 (UTC+0000)] info/network       TCP     | Shutting down the ConnectionManager
[2025-03-05 00:20:33.567 (UTC+0000)] info/eventloop     The EventLoop has stopped
[2025-03-05 00:20:33.568 (UTC+0000)] info/userland      In StartOPCLDSServer thread: argc=2, argv=127.0.0.1

[2025-03-05 00:20:33.568 (UTC+0000)] info/userland      starting LDS server...
[2025-03-05 00:20:33.573 (UTC+0000)] info/userland      LDS_StartServer.c : cannot find /usr/local/ssl/certs/ldscert44.pem
[2025-03-05 00:20:33.573 (UTC+0000)] info/userland      OPCUA LDS Server was unexpectedly shut down
Segmentation fault (core dumped)

my cmake options are :
RUN cmake -S .. -DOPENSSL_CRYPTO_LIBRARY=/usr/local/ssl/lib64/libcrypto.so -DOPENSSL_INCLUDE_DIR=/usr/local/ssl/include -DOPENSSL_SSL_LIBRARY=/usr/local/ssl/lib64/libssl.so -DUA_ARCHITECTURE=posix -DUA_DEBUG_FILE_LINE_INFO=ON -DUA_ENABLE_AMALGAMATION=OFF -DBUILD_SHARED_LIBS=OFF -DUA_ENABLE_DA=ON -DUA_ENABLE_DATATYPES_ALL=ON -DUA_ENABLE_DEBUG_SANITIZER=ON -DUA_ENABLE_DIAGNOSTICS=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_DISCOVERY_SEMAPHORE=ON -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_MQTT=ON -DUA_ENABLE_NODEMANAGEMENT=ON -DUA_ENABLE_NODESETLOADER=OFF -DUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS=ON -DUA_ENABLE_PARSING=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ENCRYPTION=ON -DUA_ENABLE_PUBSUB_FILE_CONFIG=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_TYPEDESCRIPTION=ON -DUA_ENABLE_XML_ENCODING=ON -DUA_NAMESPACE_ZERO=FULL

Please advise what does this error means.
Thanks.

Jacky

jackybek, 2025-03-05:
I have updated the Dockerfile and upload to git hub.  To facilitate testing, i also uploaded the private key and ssl certificate.
However, the steps to link these key and certificate to the correct directory in debian are not working.
Hence please manually run the following commands in ~/OPCUAProject after entering the docker container.
====================
cd /usr/local/ssl/certs/
ln -s /root/OPCUAProject/ldscert44.pem ldscert44.pem

cd /usr/local/ssl/private/
ln -s /root/OPCUAProject/ldsprivate-key.pem ldsprivate-key.pem
====================

please try and let me know what you think.
I still cannot solve the segmentation fault error.

jackybek, 2025-03-05:
The error i got is pertaining to this function: 
UA_ServerConfig_setDefaultWithSecurityPolicies(config1, 4840,              
                                                       &certificate, &privateKey,
                                                       trustList, trustListSize,
                                                       issuerList, issuerListSize,
                                                       revocationList, revocationListSize);

The error code is 0x80ab0000, which according to https://www.opcti.com/common-error-codes.aspx, it refers to "One or more arguments are invalid".

I suspect it is due to the certificate and privatekey generation process.

What is the requirement for open62541?

jackybek, 2025-03-07:
added additional commands in the Dockerfile for
1. biber
2. clang-tidy
3. latex2html
4. texlive-xetex
5. xindy

The last remaining package is cmake dll tool, which i have no idea where to find.
This will complete the setup in ccmake ..