open62541/open62541 #7149: UA_ServerConfig_setDefaultWithSecurityPolicies() function is broken

jackybek, 2025-03-06:
## Description

## Background Information / Reproduction Steps
I tested using both methods:
1. Generating my own self signed certificate and key using openssl
2. Call the UA_CreateCertificate() function provided by the library

Both of them i get this error : Bad Invalid argument
For method 2, I followed the example given.

encryptServer(UA_Server *) 
{
        UA_ByteString derPrivKey = UA_BYTESTRING_NULL;
	UA_ByteString derCert = UA_BYTESTRING_NULL;

	UA_String subject[3] = {UA_STRING_STATIC("C=SG"),
				UA_STRING_STATIC("O=Virtual Skies"),
				UA_STRING_STATIC("CN=lds.virtualskies.com.sg") };

	UA_UInt32 lenSubject = 3;
	UA_String subjectAltName[2] = {UA_STRING_STATIC("DNS.1:localhost"),
				       UA_STRING_STATIC("DNS.2:lds.virtualskies.com.sg") };
	UA_UInt32 lenSubjectAltName = 2;
	
	UA_KeyValueMap *kvm = UA_KeyValueMap_new();
	UA_UInt16 expiresIn = 3650;
	UA_KeyValueMap_setScalar(kvm, UA_QUALIFIEDNAME(0, "expires-in-days"),
				(void *)&expiresIn , &UA_TYPES[UA_TYPES_UINT16]);
	UA_UInt16 keyLength = 2048;
	UA_KeyValueMap_setScalar(kvm, UA_QUALIFIEDNAME(0, "key-size-bits"),
				(void *)&keyLength, &UA_TYPES[UA_TYPES_UINT16]);

	UA_StatusCode status = UA_CreateCertificate(
						UA_Log_Stdout, subject, lenSubject, subjectAltName, lenSubjectAltName,
						UA_CERTIFICATEFORMAT_DER, kvm, &derPrivKey, &derCert);

	UA_ServerConfig *config = UA_Server_getConfig(uaLDSServer);
	status = UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840, 
							&derCert, &derPrivKey,
							trustList, trustListSize, 
							issuerList, issuerListSize, 
							revocationList, revocationListSize);
}

In my main program
main()
{
	UA_Server *uaLDSServer = UA_Server_new();
     	UA_ServerConfig *config = UA_Server_getConfig(uaLDSServer);
     	UA_ServerConfig_setMinimal(config, 4840, NULL);

        encryptServer(uaLDSServer);
}

Used CMake options:
-DOPENSSL_CRYPTO_LIBRARY=/usr/local/ssl/lib64/libcrypto.so -DOPENSSL_INCLUDE_DIR=/usr/local/ssl/include -DOPENSSL_SSL_LIBRARY=/usr/local/ssl/lib64/libssl.so -DUA_ARCHITECTURE=posix -DUA_DEBUG_FILE_LINE_INFO=ON -DUA_ENABLE_AMALGAMATION=OFF -DBUILD_SHARED_LIBS=OFF -DUA_ENABLE_DA=ON -DUA_ENABLE_DATATYPES_ALL=ON -DUA_ENABLE_DEBUG_SANITIZER=ON -DUA_ENABLE_DIAGNOSTICS=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_DISCOVERY_SEMAPHORE=ON -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_MQTT=ON -DUA_ENABLE_NODEMANAGEMENT=ON -DUA_ENABLE_NODESETLOADER=OFF -DUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS=ON -DUA_ENABLE_PARSING=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ENCRYPTION=ON -DUA_ENABLE_PUBSUB_FILE_CONFIG=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_TYPEDESCRIPTION=ON -DUA_ENABLE_XML_ENCODING=ON -DUA_NAMESPACE_ZERO=FULL

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): v1.4.9
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: linux x86
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X ] Critical issue

jpfr, 2025-03-06:
What is the return value of UA_CreateCertificate?

jackybek, 2025-03-06:
The return value is GOOD.

jpfr, 2025-03-06:
We need to add more logging.

In the meantime, you can use a debugger to step through the function.
Or add printf-debugging to see where the error is coming from.

The code you posted looks good.
But we don't see where the `trustList` and other variables are coming from.

jackybek, 2025-03-06:
My codes are in github/ jackybek / LDS-44
The program that crashed is Lds_encryptServer.c

jackybek, 2025-03-06:
i trace the the codes in v1.4.8 and added debug statements.

in main(), i have these variables declared and initialised:
{
        UA_Server *uaLDSServer = UA_Server_new();
        UA_ServerConfig *config = UA_Server_getConfig(uaLDSServer);
        UA_ServerConfig_setDefault(config);

        **if (!config)
                  exit(-1);**
        int status = encryptServer(uaLDSServer);
}

In encryptServer()
{
Before I call UA_ServerConfig_setDefaultWithSecurityPolicies(), i checked the following:

 UA_ServerConfig *config = UA_Server_getConfig(uaLDSServer);
i f (!config) {--print error)}
else call UA_ServerConfig_setDefaultWithSecurityPolicies().

In UA_ServerConfig_setDefaultWithSecurityPolicies(), [ua_config_default.c], it calls UA_StatusCode retval = setDefaultConfig(conf, portNumber);
It does not return UA_STATUSCODE_GOOD.

I traced further into setDefaultConfig() and it fails at this statement:
if (!conf) return UA_STATUSCODE_BADINVALIDARGUMENT;

All throughout from main() to encryptServer() to generateSSCert(), I passed the pointer of uaLDSServer between these 3 functions.
The weird thing is that right from the begining in main(), it already failed at : if (!config) check.

jackybek, 2025-03-11:
I looked through the codes and found this : 
UA_ServerConfig *UA_Server_getConfig(UA_Server *server) {
UA_CHECK_MEM(server, return NULL);
return &server->config;
}

I suspect UA_CHECK_MEM() is not correct, that's why it returns a NULL.

jackybek, 2025-03-15:
i updated my own copy of config.h.in
# define UA_UNLIKELY (x) !x

The issue is resolved.

jackybek, 2025-03-15:
I updated my own copy of this <config.h.in>
    # define UA_UNLIKELY(x) !x

The previous error is resolved i,e these 2 functions worked now:
UA_ServerConfig_setMinimal(config, 4840, NULL);
status = UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840,
                                                        derCert, derPrivKey,
                                                        trustList, trustListSize,
                                                        issuerList, issuerListSize,
                                                        revocationList, revocationListSize);

closed 2025-03-15 by jackybek