open62541/open62541 #7151: Cannot build on MacOS for MacOS

Gr3yShad0w, 2025-03-07:
== Steps to reproduce

I'm trying to build open62541 for MacOS on a MacOS.

CMake configs:

```
-- open62541 Version: v1.4.10-1127-gd451c75d5
-- The selected architecture is: posix
-- Configuring done (0.9s)
-- Generating done (0.1s)
-- Build files have been written to: /tmp/open62541/build
-- Cache values
BUILD_SHARED_LIBS:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_OSX_ARCHITECTURES:STRING=arm64
CMAKE_OSX_DEPLOYMENT_TARGET:STRING=13.0
CMAKE_OSX_SYSROOT:PATH=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX15.2.sdk
UA_ARCHITECTURE:STRING=none
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DATATYPES_ALL:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_GDS_PUSHMANAGEMENT:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODESETLOADER:BOOL=OFF
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_XML_ENCODING:BOOL=OFF
UA_LOGLEVEL:STRING=100
UA_MULTITHREADING:STRING=100
UA_NAMESPACE_ZERO:STRING=REDUCED
UA_NODESET_DIR:STRING=/tmp/open62541/deps/ua-nodeset
```

When I then try to use the given `.a` file in combination with the `.h` file in a project I get the following errors:

```
Undefined symbols for architecture arm64:
  "_UA_ConnectionManager_new_POSIX_TCP", referenced from:
      _setDefaultConfig in libopen62541.a[74](ua_config_default.c.o)
      _UA_ClientConfig_setDefault in libopen62541.a[74](ua_config_default.c.o)
  "_UA_ConnectionManager_new_POSIX_UDP", referenced from:
      _setDefaultConfig in libopen62541.a[74](ua_config_default.c.o)
      _UA_ClientConfig_setDefault in libopen62541.a[74](ua_config_default.c.o)
  "_UA_DateTime_localTimeUtcOffset", referenced from:
      _UA_Log_Stdout_log in libopen62541.a[70](ua_log_stdout.c.o)
  "_UA_DateTime_now", referenced from:
      _UA_random_seed in libopen62541.a[7](ua_util.c.o)
      _UA_Server_init in libopen62541.a[12](ua_server.c.o)
      _UA_DataSetReader_generateDataSetMessage in libopen62541.a[36](ua_pubsub_readergroup.c.o)
      _UA_Log_Stdout_log in libopen62541.a[70](ua_log_stdout.c.o)
      _setDefaultConfig in libopen62541.a[74](ua_config_default.c.o)
  "_UA_DateTime_nowMonotonic", referenced from:
      _UA_PubSubConnection_decodeNetworkMessage in libopen62541.a[31](ua_pubsub_connection.c.o)
      _UA_PubSubConnection_process in libopen62541.a[31](ua_pubsub_connection.c.o)
      _interruptServer in libopen62541.a[74](ua_config_default.c.o)
  "_UA_EventLoop_new_POSIX", referenced from:
      _setDefaultConfig in libopen62541.a[74](ua_config_default.c.o)
      _UA_ClientConfig_setDefault in libopen62541.a[74](ua_config_default.c.o)
  "_UA_InterruptManager_new_POSIX", referenced from:
      _setDefaultConfig in libopen62541.a[74](ua_config_default.c.o)
ld: symbol(s) not found for architecture arm64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
```

Note that I used `UA_ARCHITECTURE=none` and not `UA_ARCHITECTURE=posix` because when I try to build it with that configuration I get

```
[  4%] Built target open62541-amalgamation
[  5%] Built target open62541-generator-types
[  7%] Built target open62541-generator-namespace
[  7%] Built target open62541-generator-transport
[  8%] Built target open62541-generator-statuscode
[  8%] Built target open62541-code-generation
[ 65%] Built target open62541-object
[ 66%] Building C object CMakeFiles/open62541-plugins.dir/arch/posix/eventloop_posix.c.o
In file included from /tmp/open62541/arch/posix/eventloop_posix.c:9:
/tmp/open62541/arch/posix/eventloop_posix.h:266:10: fatal error: 'sys/inotify.h' file not found
  266 | #include <sys/inotify.h>
      |          ^~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/open62541-plugins.dir/arch/posix/eventloop_posix.c.o] Error 1
make[1]: *** [CMakeFiles/open62541-plugins.dir/all] Error 2
make: *** [all] Error 2
```

From what I figured out, was that `inotify` is Linux specific and not posix specific thats the reason it's failing.
Correct me if I'm wrong but in my mind, Linux is not posix conform while MacOS is.

I also tried to directly used the .c and .h file generated from the amalgamation but then the compilation of the c file fails eventho I am using C99 as the C-Version:

```
.../ThirdParty/Open62541/open62541.c:33181:9: error: cannot jump from this goto statement to its label
.../ThirdParty/Open62541/open62541.c:33176:9: error: cannot jump from this goto statement to its label
.../ThirdParty/Open62541/open62541.c:33171:9: error: cannot jump from this goto statement to its label
.../ThirdParty/Open62541/open62541.c:38197:13: error: cannot jump from this goto statement to its label

.... (multiple of these)
```

== Conclusion

My question is:

How can I generate a static library for MacOS on a MacOS, which settings do I have to change?

jpfr, 2025-03-11:
We need to remove the inotify support for macOS.
The arch=none build requires you to add the arch-dependent eventloop parts.

And we need a macOS GitHub runner for the CI.

Gr3yShad0w, 2025-03-14:
@jpfr so do I understand correctly that currently, there is no way to generate a static library that works on ARM OSX?

jpfr, 2025-04-20:
Or course it’s possible.
We now also have automated tests for MacOS.
Select the POSIX architecture.

jpfr, 2025-04-20:
Try the current master branch.

closed 2025-04-20 by jpfr