open62541/open62541 #7155: Simple test to proof that something is broken in the open62541 build process

jackybek, 2025-03-07:
## Description
I wrote a simple program(LDS_main.c) that does the following:

        UA_Server *uaLDSServer = UA_Server_new();
        UA_ServerConfig *config = UA_Server_getConfig(uaLDSServer);
        **UA_ServerConfig_clear(config);**
        UA_ServerConfig_setMinimal(config, 4840, NULL);

I then check my copy of the CMakeLists.txt on my computer.
It has this line : ${PROJECT_SOURCE_DIR}/src/server/ua_server_config.c

I then perform the following:
cmake ..
ccmake ..
make

I then build my application, but I get 
home/pi/OPCUAProject/LDS_main.c:111: undefined reference to `UA_ServerConfig_clear'

Somehow this function is not included in the libopen62541.a library when i did the build process.

Besides using cmake, what other options do i have to build the libopen62541.a library properly?

## Background Information / Reproduction Steps

==============================
This is my Makefile to build the simple app.

HOME=/home/pi/
CC=gcc
CFLAGS1=-g -std=c99 -Wall -Werror -Wno-implicit -Wfatal-errors
#CC=g++
#CFLAGS1=-x c++ -std=c++98 -v -fpermissive -Wno-literal-suffix -Wno-write-strings -Wno-long-long -Wno-return-type
CFLAGS2=-I$(HOME) -I$(HOME)include/ -I$(HOME)plugins/ -I$(HOME)src/ -I$(HOME)open62541/build/src_generated/ \
-I$(HOME)arch/ -I$(HOME)deps/ -I$(HOME)plugins/include/ -I/usr/local/include/ -I$(HOME)open62541/plugins/ -I$(HOME)open62541/include/ \
-I$(HOME)open62541/plugins/include/
CFLAGS3=-g -pass-exit-codes
DEPS=
LIBS=-lm -lrt -lpthread  -lcrypto -lssl -lmbedcrypto -lmbedtls -lmbedx509 -lwebsockets
#OBJ=LDS_mainServer.o LDS_StartServer.o
OBJ= LDS_Misc.o LDS_configureServer.o LDS_encryptServer.o LDS_main.o
ODIR=obj
LDIR1=-L/usr/local/lib/
LDIR2=-L/usr/lib/
LDFLAGS=-L$(HOME)open62541/build/bin/ -l:libopen62541.a -l:liblibmdnsd.a
all: myNewLDSServer
%.o: %.c $(DEPS)
        $(CC) $(CFLAGS1) $(CFLAGS2) $(CFLAGS3) -c  $< -o $@
myNewLDSServer: $(OBJ)
        $(CC) $(LDIR1) $(LDIR2) -o $@ $^ $(LDFLAGS) $(LIBS)

=========================

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag): 1.4.9
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: debian bullseye
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ X] Critical issue

jpfr, 2025-03-11:
This should be _clean instead of _clear.

https://github.com/open62541/open62541/blob/cdb5fa035b1a123703fe84cec29b22057eadbf72/include/open62541/server.h#L398

It has been renamed on master to be consistent.
We should add a macro to allow both.

Are you making a PR for the 1.4 branch?

jackybek, 2025-03-16:
After I downloaded open62541.c and open62541.h and compile from here (without using git clone), all the errors disappeared.

closed 2025-03-16 by jackybek