open62541/open62541 #7162: Potential Regression in 1.3.15 to 1.3.8: UA_NodeId_parse - OPAQUE mode strings

Flow86, 2025-03-14:
## Description

Potential regression in parsing node ids in 1.3.15 to 1.3.8

## Background Information / Reproduction Steps

We updated our open62541 from 1.3.8 to 1.3.15 as a pre-step to upgrade to 1.4
We got an error "BadInternalError" in our code
We condensed it down to `UA_NodeId_parse`:

Example program:
```
#include <open62541/client.h>

#include <string>
#include <cassert>
#include <iostream>

int main(void)
{
    std::string input = "ns=12;b=1234567890";
    auto* node = UA_NodeId_new();
    const auto ret = UA_NodeId_parse(node, UA_STRING(const_cast<char*>(input.data())));

    std::cerr << "input: " << input << "\n";
    std::cerr << "ret: " <<  UA_StatusCode_name(ret) << "\n";
    assert(ret == UA_STATUSCODE_GOOD);
}
```

once linked against 1.3.8 and once against 1.3.15, built with Visual Studio:

```
cl regression.cpp /LINK /LIBPATH=open62541\lib open62541.lib
```

Used CMake options to build the libraries:

```bash
cmake -DUA_ENABLE_ENCRYPTION=OPENSSL -DBUILD_SHARED_LIBS=ON -DOPENSSL_ROOT_DIR=openssl-3.x
```

Output with 1.3.8:
```
input: ns=12;b=1234567890
ret: Good
```

Output with 1.3.15:
```
input: ns=12;b=1234567890
ret: BadInternalError
Assertion failed: ret == UA_STATUSCODE_GOOD, file .\regression.cpp, line 15
```

We also tried other OPAQUE-mode ByteStrings, all of them get rejected, besides the empty string:

```
std::string input = "ns=12;b=";
```

output:
```
input: ns=12;b=
ret: Good
```

Are we doing something wrong there, or is it a regression?

Flow86, 2025-03-14:
Ah. After investigating further (building the two dlls with debug symbols) and digging deeper into the library,
I've realized that open62541 expects a ByteString to be base64 encoded and was broken in 1.3.8 when using invalid base64 input.
Our application only checked for "oh its ByteString, we do not support that" and of course now the stuff expects valid base64.

But in general: enforcing base64 encoding is not according to the OPC UA specs at https://reference.opcfoundation.org/Core/Part6/v105/docs/5.2.2 ?

jpfr, 2025-03-23:
Base64 is enforced: https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.12#_Ref122651839

Note that the human readable encoding is different from the binary encoding.

closed 2025-03-23 by jpfr