open62541/open62541 #7178: Segmentation fault on client connect using certificates

Schokoladenblau, 2025-03-20:
Hello, I have problems to connect to a OPC-UA server using certificates. I somehow always endup with an segementation fault and have no Idea why. I am using the version 1.4.6 within a C++ application running in a docker container. The code:

```
 /* Most of the following code is adapted*/
  /* Load certificate and private key */
  UA_ByteString certificate = loadFile(m_certificate_path.c_str());
  UA_ByteString privateKey  = loadFile(m_private_key_path.c_str());

  UA_STACKARRAY(UA_ByteString, trustList, m_trust_list_paths.size());
  for(size_t trustListCount = 0; trustListCount < m_trust_list_paths.size(); trustListCount++)
    trustList[trustListCount] = loadFile(m_trust_list_paths[trustListCount].c_str());

  /* Revocation lists are supported, but not used for the example here */
  UA_ByteString *revocationList = NULL;
  size_t revocationListSize = 0;

  if(m_client != nullptr) 
    UA_Client_delete(m_client);

  m_client = UA_Client_new();

  UA_ClientConfig *cc = UA_Client_getConfig(m_client);
  cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
  UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey,
                                      trustList, m_trust_list_paths.size(),
                                      revocationList, revocationListSize
  );

  cc->timeout = 5 * 1000;

  /* Secure client connect */
  cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT; /* require encryption */
  cc->securityPoliciesSize = 2;

  cc->securityPolicies = new UA_SecurityPolicy[cc->securityPoliciesSize];
	UA_SecurityPolicy_None(&cc->securityPolicies[0], UA_BYTESTRING_NULL, cc->logging);

	//Configure a secure connection with Basic256Sha256
	cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
	UA_SecurityPolicy_Basic256Sha256(&cc->securityPolicies[1], certificate, privateKey, cc->logging);

  m_connection_string = "opc.tcp://" + m_config.ip + ":" + std::to_string(m_config.ua_port);
  
  // ConnectAsync to the server
  UA_StatusCode retval = UA_Client_connect(m_client, m_connection_string.c_str());
  if(retval != UA_STATUSCODE_GOOD){
    m_connection_status.SetState(ConnectionStatus::kError, m_processing_job_context, "Connecting to PLC failed", false);
    SPDLOG_INFO("Connecting to PLC {} failed with error: {}", m_connection_string, UA_StatusCode_name(retval));
    return;
  }
```

The logs are as follows:
```
info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionClosed
error/securitypolicy       Could not create securityContext: BadSecurityChecksFailed
info/eventloop     Starting the EventLoop
warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
info/network       TCP 28  | Opening a connection to "192.168.113.230" on port 4840
info/channel       TCP 28  | SC 2443445016 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
info/client        Rejecting endpoint 0: security mode does not match
info/client        Rejecting endpoint 1: security mode does not match
info/client        Rejecting endpoint 2: security policy does not match the configuration
info/client        Rejecting endpoint 3: security mode does not match
info/client        Selected endpoint with EndpointUrl opc.tcp://192.168.113.230:4840, SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
info/channel       TCP 28  | SC 2443445016 | SecureChannel closed
warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
info/network       TCP 29  | Opening a connection to "192.168.113.230" on port 4840
info/network       TCP 28  | Socket closed
```

the debugger indicates that the error occured in open62541/plugins/crypto/mbedtls/ua_securitypolicy_basic256sha256.c : 392
```
int mbedErr = mbedtls_ctr_drbg_random(&pc->drbgContext, out->data, out->length);
```

I would really appriciate some help - could also just be a good documented example of how to use certificates with this library. The examples provided lack explenations and seem to not work (the above code is copied from there for the most part)

Schokoladenblau, 2025-03-31:
Thanks for your reply!

I did now find the example I was looking for : examples/client_connect
 -> This worked for me and has good comments 😄
I previously only looked at the examples under examples/encryption/
-> They might need some adjustments ?

I did not investigate the segmentation fault further but I think it came from some configuration error on my end. 

I would also suggest to implement more debug log messages for the configuration-checking mechanisms. The message "Could not create securityContext: BadSecurityChecksFailed" is not very meaningful. 
I did also notice that the connection procedure does not produce any log messages with log level debug. I think the connection part is the one where most beginners struggel and more detailed messages might make sense here.

jackybek, 2025-04-09:
You can try adding the server ssl certificate to the client' trustlist and load it in your program.
Similarly, you can try adding the client's ssl certificate to the server's trustlist and load it.