open62541/open62541 #7179: ReverseConnect example : does not seem to work

jackybek, 2025-03-21:
## Description

## Background Information / Reproduction Steps

My OPC server instance is running on opc.tcp://192.168.1.44:4840.
My UAExpert instance is running on host 192.168.1.157 (laptop).
Without using Reverse Connect, UAExpert can connect to the OPC Server using the certificates provided by the Server.

I then configure 'Reverse Connect' option in UAExpert to use port 4839.
I also open the firewall on the laptop to allow all inbound connection.

[image]

[image]

In my code, the invoke sequence is as follows:
1. create a new UA_Server instance using UA_Server_new()
2. set default configuration using UA_ServerConfig_setDefault()
3. encrypt the OPC Server
4. configure additional parameters
5. call UA_Server_run_startup() to start OPC Server
6. call reverseConnect() function
7. register to a OPCUA Local Directory Server
8. create Methods, Events, MonitoredItems, Historian
9. configure pubsub UADP, ETH and MQTT
10. at the final segment of my program, i have

```
while (running) 
       UA_Server_run_iterate(server, true); 
if (!running)
                {
                        if (lds != NULL)
                        {
                                int status = UA_Server_deregisterDiscovery(server, lds, UA_STRING(argv[3]));
                                if (status != UA_STATUSCODE_GOOD)
                                        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                                                        "--------SV_main.c : Could not unregister OPCUA server from discovery server. StatusCode %s",
                                                        UA_StatusCode_name(status));

                                UA_Server_run_shutdown(server);
                                UA_Server_delete(server);
                        }
                }
```

In my ReverseConnect function, i basically copied everything from github, only changing the reverse_connection port to 4839 and the ip to point to my laptop (192.168.1.157), where UAExpert client program is running.  This is the output logs showing the reverseconnect -
```
[2025-03-21 16:57:40.334 (UTC+0800)] info/userland      ==========================================================
[2025-03-21 16:57:40.334 (UTC+0800)] info/userland      --------SV_reverseConnect.c : entering reverseConnect()
[2025-03-21 16:57:40.335 (UTC+0800)] info/userland      --------SV_reverseConnect.c : successfully assembled SVR endpoint url : <opc.tcp://192.168.1.157:4839>
[2025-03-21 16:57:40.335 (UTC+0800)] info/network       TCP 6   | Opening a connection to "192.168.1.157" on port 4839
[2025-03-21 16:57:40.336 (UTC+0800)] info/userland      Reverse connect state callback for 1 with context 0x1e240: State 2
[2025-03-21 16:57:40.337 (UTC+0800)] info/userland      --------SV_reverseConnect.c : add ReverseConnect success
[2025-03-21 16:57:40.337 (UTC+0800)] info/userland      ==========================================================
```
Other logs pertaining to steps 7,8 and 9 are output in between.  After the above. I start getting these messages:
```
[2025-03-21 16:57:50.479 (UTC+0800)] info/userland      Reverse connect state callback for 1 with context 0x1e240: State 5
[2025-03-21 16:58:01.478 (UTC+0800)] info/channel       TCP 6   | SC 1  | SecureChannel has timed out
[2025-03-21 16:58:01.479 (UTC+0800)] info/channel       TCP 6   | SC 1  | SecureChannel closed due to timeout
[2025-03-21 16:58:01.479 (UTC+0800)] info/userland      Reverse connect state callback for 1 with context 0x1e240: State 2
[2025-03-21 16:58:01.479 (UTC+0800)] info/network       TCP 6   | Socket closed
[2025-03-21 16:58:10.335 (UTC+0800)] info/server        Attempt to reverse reconnect to 192.168.1.157:4839
[2025-03-21 16:58:10.336 (UTC+0800)] info/network       TCP 6   | Opening a connection to "192.168.1.157" on port 4839
[2025-03-21 16:58:10.336 (UTC+0800)] info/userland      Reverse connect state callback for 1 with context 0x1e240: State 5
[2025-03-21 16:58:10.338 (UTC+0800)] info/channel       TCP 6   | SC 2  | SecureChannel closed
[2025-03-21 16:58:10.338 (UTC+0800)] info/userland      Reverse connect state callback for 1 with context 0x1e240: State 2
[2025-03-21 16:58:10.338 (UTC+0800)] info/network       TCP 6   | Socket closed
```

On the UAExpert log screen, i get these errors:

[image]

Used CMake options: No cmake options, using open62541.c and open62541.h (almagamation) downloaded from github.

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): 1.4.10
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: debian 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X ] Self-contained code example attached
 - [ ] Critical issue

jackybek, 2025-08-16:
here is my code to implement reverse connect.

[SV_ReverseConnect.txt](https://github.com/user-attachments/files/21808337/SV_ReverseConnect.txt)

There is this parameter (void *)123456 passed into UA_Server_addReverseConnect() which i don't understand what is the significance of it.
Could this be the reason for the errors described above?

udovichenkoAlexander, 2025-10-31:
`Client Endpoint` in UAExpert mast be equal to that server send in RHE. By Default open62541 usine `gethostname` API for this.

You can check RHE packet with wireshark (decode as `OpcUa`)

[image]

udovichenkoAlexander, 2025-11-20:
When you configure UaExpert for reverse Connect then `EndPointUrl` it is not for connect, it's just string. this string must be equal to `EndPointUrl` in `RHE`. `RHE` - first packet, that server send to client when do reverse connect.

Client can listen mulitiple servers on same port. With `EndPointUrl` it can separate two diferent servers.

So look what `EndPointUrl` in `RHE` (with wireshark like on my screenshot) and ensure that it equal to `EndPointUrl` in UAExpert.

jackybek, 2025-11-28:
I turned off encryption in my code.
I also checked my source code that launches reverseconnect.

        int retval = UA_Server_addReverseConnect(uaServer, UA_STRING(client_endpoint),
                        reverseConnectStateCallback, (void *)callbackContext, &handle);

Here the client_endpoint i entered is opc.tcp:192.168.1.157:4839 (my laptop),  which matches uaExpert settings.

But you said i need to look for the RHE message in wireshark to look for a string to enter as the client endpoint in uaExpert.
So does it mean that in my program, I should not be using opc.tcp://192.168.1.157:4839, but opc.tcp://<string>:4839?

Then I checked the source code of addReverseConnect() in the repo.  

[image]

It seemed to infer that i should be using the server url, instead of the client url.

jackybek, 2025-11-28:
After reviewing the settings and code and making the necessary amendments, this is the current state:

Source Code: use the client URL in UA_Server_addReverseConnect() function.
Matrikon - ReverseConnect Settings
Server endpoint opc.tcp://192.168.1.109:4840 (opc ua server)
client listening port: 4839 (set in the source code)

 I get message " Server:opc.tcp://OPCSvr-109:4840 initiated reverse connection in Matrikon

In my oPUServer , I get message 
~~~
Attempt to reverse reconnect to 192.168.1.157:4839
[2025-11-28 09:38:44.130 (UTC+0800)] info/network       TCP 7   | Opening a connection to "192.168.1.157" on port 4839
[2025-11-28 09:38:44.131 (UTC+0800)] info/userland      Entering reverseConnectStateCallback() in SV_ReverseConnect.c
[2025-11-28 09:38:44.131 (UTC+0800)] info/userland      --------SV_ReverseConnect.c : Reverse connect state callback for handle (1) with context (0x1e240) : State (5)
[2025-11-28 09:38:44.143 (UTC+0800)] info/userland      Entering reverseConnectStateCallback() in SV_ReverseConnect.c
[2025-11-28 09:38:44.143 (UTC+0800)] info/userland      --------SV_ReverseConnect.c : Reverse connect state callback for handle (1) with context (0x1e240) : State (8)
[2025-11-28 09:38:54.619 (UTC+0800)] info/channel       TCP 7   | SC 8  | SecureChannel has timed out
[2025-11-28 09:38:54.621 (UTC+0800)] info/channel       TCP 7   | SC 8  | SecureChannel closed due to timeout

jackybek, 2025-11-28:
I changed UA_RuleHandling verifyRequestTimestamp = UA_RULEHANDLING_ACCEPT, but still get "The server sends no timestamp in the request header".

Is there another setting in UA_Config that specify sendTimeStamp?

jackybek, 2025-11-29:
Can I check whether the following behaviour is correct:

Scenario 1
1. Disable the reverseconnect module in my code (i.e. dont run)
2. Connect normally using uaExpert and Matrikon using Aes256Sha256 - SUCCESS
3. Note there are 2 active client sessions now.

Scenario 2
1. Enable reverseconnect module 
2. Connect via Matrikon reverse Connect method using Aes256Sha256 - SUCCESS
3. Try to connect uaExpert normally to OPCServer - FAIL - certificate error

If the above behaviour is correct, does it mean that OPCUA server can only accept 1 kind of connections, either normal client-initiated connection OR reverse-connect but NOT Both?

Please advise. Thanks.
(Strangely yesterday I was able to do reverse-connect using Matrikon and normal-connect using uaExpert.  But today this failed).