open62541/open62541 #7201: mqtt related errors when making v.1.4.11

jackybek, 2025-04-07:
## Description
While build the open62541 source codes, i get mqtt-related errors.

I used the following commands to get the source codes from github:

sudo git clone https://github.com/open62541/open62541.git --branch v1.4.11 -c advice.detachedHead=FALSE
cd op62541
sudo git submodule update --init --recursive

``
[ 66%] Building C object CMakeFiles/open62541-object.dir/open62541.c.o
/home/pi/open62541.v1.4.11git/build/open62541.c:117273:24: error: field ‘client’ has incomplete type
117273 |     struct mqtt_client client;
       |                        ^~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c: In function ‘mqtt_pal_sendall’:
/home/pi/open62541.v1.4.11git/build/open62541.c:117324:16: error: ‘MQTT_ERROR_SOCKET_ERROR’ undeclared (first use in this function)
117324 |         return MQTT_ERROR_SOCKET_ERROR;
       |                ^~~~~~~~~~~~~~~~~~~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c:117324:16: note: each undeclared identifier is reported only once for each function it appears in
/home/pi/open62541.v1.4.11git/build/open62541.c: In function ‘shutdownBrokerConnection’:
/home/pi/open62541.v1.4.11git/build/open62541.c:117397:9: error: implicit declaration of function ‘mqtt_disconnect’ [-Wimplicit-function-declaration]
117397 |         mqtt_disconnect(&bc->client);
       |         ^~~~~~~~~~~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c:117397:9: warning: nested extern declaration of ‘mqtt_disconnect’ [-Wnested-externs]
/home/pi/open62541.v1.4.11git/build/open62541.c:117398:9: error: implicit declaration of function ‘__mqtt_send’ [-Wimplicit-function-declaration]
117398 |         __mqtt_send(&bc->client);
       |         ^~~~~~~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c:117398:9: warning: nested extern declaration of ‘__mqtt_send’ [-Wnested-externs]
/home/pi/open62541.v1.4.11git/build/open62541.c: In function ‘removeTopicConnection’:
/home/pi/open62541.v1.4.11git/build/open62541.c:117470:9: error: implicit declaration of function ‘mqtt_unsubscribe’ [-Wimplicit-function-declaration]
117470 |         mqtt_unsubscribe(&bc->client, (const char*)tc->topic.data);
       |         ^~~~~~~~~~~~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c:117470:9: warning: nested extern declaration of ‘mqtt_unsubscribe’ [-Wnested-externs]
/home/pi/open62541.v1.4.11git/build/open62541.c: In function ‘MQTTKeepAliveCallback’:
/home/pi/open62541.v1.4.11git/build/open62541.c:117595:5: error: implicit declaration of function ‘mqtt_ping’ [-Wimplicit-function-declaration]
117595 |     mqtt_ping(&bc->client);
       |     ^~~~~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c:117595:5: warning: nested extern declaration of ‘mqtt_ping’ [-Wnested-externs]
/home/pi/open62541.v1.4.11git/build/open62541.c: At top level:
/home/pi/open62541.v1.4.11git/build/open62541.c:117600:50: warning: ‘struct mqtt_response_publish’ declared inside parameter list will not be visible outside of                                                                              this definition or declaration
117600 | MQTTPublishResponseCallback(void** state, struct mqtt_response_publish *publish) {
       |                                                  ^~~~~~~~~~~~~~~~~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c: In function ‘MQTTPublishResponseCallback’:
/home/pi/open62541.v1.4.11git/build/open62541.c:117604:31: error: invalid use of undefined type ‘struct mqtt_response_publish’
117604 |     UA_String topic = {publish->topic_name_size,
       |                               ^~
/home/pi/open62541.v1.4.11git/build/open62541.c:117605:52: error: invalid use of undefined type ‘struct mqtt_response_publish’
117605 |                        (UA_Byte*)(uintptr_t)publish->topic_name};
       |                                                    ^~
/home/pi/open62541.v1.4.11git/build/open62541.c:117606:33: error: invalid use of undefined type ‘struct mqtt_response_publish’
117606 |     UA_ByteString msg = {publish->application_message_size,
       |                                 ^~
/home/pi/open62541.v1.4.11git/build/open62541.c:117607:54: error: invalid use of undefined type ‘struct mqtt_response_publish’
117607 |                          (UA_Byte*)(uintptr_t)publish->application_message};
       |                                                      ^~
/home/pi/open62541.v1.4.11git/build/open62541.c: In function ‘MQTTNetworkCallback’:
/home/pi/open62541.v1.4.11git/build/open62541.c:117682:9: error: implicit declaration of function ‘mqtt_init’; did you mean ‘Sort_init’? [-Wimplicit-function-de                                                                             claration]
117682 |         mqtt_init(&bc->client, bc,
       |         ^~~~~~~~~
       |         Sort_init
/home/pi/open62541.v1.4.11git/build/open62541.c:117682:9: warning: nested extern declaration of ‘mqtt_init’ [-Wnested-externs]
/home/pi/open62541.v1.4.11git/build/open62541.c:117688:14: warning: ISO C forbids forward references to ‘enum’ types [-Wpedantic]
117688 |         enum MQTTErrors err = mqtt_connect(&bc->client, NULL, NULL, NULL, 0,
       |              ^~~~~~~~~~
/home/pi/open62541.v1.4.11git/build/open62541.c:117688:14: error: variable ‘err’ has initializer but incomplete type
/home/pi/open62541.v1.4.11git/build/open62541.c:117688:31: error: implicit declaration of function ‘mqtt_connect’; did you mean ‘BIO_connect’? [-Wimplicit-funct                                                                             ion-declaration]
117688 |         enum MQTTErrors err = mqtt_connect(&bc->client, NULL, NULL, NULL, 0,
       |                               ^~~~~~~~~~~~
       |                               BIO_connect
/home/pi/open62541.v1.4.11git/build/open62541.c:117688:31: warning: nested extern declaration of ‘mqtt_connect’ [-Wnested-externs]
[... 258 lines cut ...]
UA_MULTITHREADING:STRING=100
UA_NAMESPACE_ZERO:STRING=FULL
UA_NODESET_DIR:STRING=/home/pi/open62541.v1.4.11git/deps/ua-nodeset
XELATEX_COMPILER:FILEPATH=/usr/bin/xelatex
XINDY_COMPILER:FILEPATH=/usr/bin/xindy
pkgcfg_lib_PC_LIBXML_xml2:FILEPATH=/usr/lib/aarch64-linux-gnu/libxml2.so
pkgcfg_lib__OPENSSL_crypto:FILEPATH=/usr/lib/aarch64-linux-gnu/libcrypto.so
pkgcfg_lib__OPENSSL_ssl:FILEPATH=/usr/lib/aarch64-linux-gnu/libssl.so
``

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): 1.4.11
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: debian bookworm arm64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X ] Critical issue

jackybek, 2025-04-07:
if i turn off almagamation, the errors disappeared.
So please review the part on why make fails when producing single open62541.c and open62541.h

jackybek, 2025-04-08:
I am able to repeat the issue so please advise how I can go about generating the single file open62541.c and open62541.h and used it in my Makefile without errors.

**version : 1.4.11**
1. source : git clone https://github.com/open62541/open62541.git --branch v1.4.11 -c advice.detachedHead=FALSE
2 sudo git submodule update --init --recursive
3. During make, errors related to MQTT (almagamation=ON), but open62541.h is produced (along with open62541.c)
4. Next, set: almagamation=OFF, discovery_multicast=ON, make = passed
5. libopen62541.a is produced
6. use libopen62541.a and open62541.h in application Makefile and application source codes (but don't use open62541.c in Makefile)
7. Successfully build application

As you can see, the above steps does not allow me to use the open62541.c (instead i have to rely on libopen62541.a in the Makefile).
If i remove the libopen62541.a from the Makefile and add in open62541.c as one of the source file for compilation, i will get compilation errors, which is weird.

Because of this issue, I am not able to debug my program to resolve runtime errors such as this: Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: **BadInternalError**

There are tons of 'BadInternalError' spread all over the entire open62541.c and without adding in printf statements in open62541.c, it is impossible to debug the error.

jackybek, 2025-04-08:
I tried another way.  this time i downloaded open62541-1.4.11.1.tar.gz from github and follow the instructions below:

1/2.. sudo cmake .. && sudo ccmake ..
3. set almagamation = OFF first, configure and generate
4. sudo make => libopen62541.a is produced
5. then change almagamation = ON, libopen62541.a, open62541.c and open62541.h is produced
6. copy open62541.h and .c to ~/OPCProject
7. include in Makefile open62541.c and exclude libopen62541.a
8. go to open62541.h and enable the desired #define
9. sudo make => link errors:
	a. open62541.c:6839:5: error: unknown type name ‘mdns_daemon_t’
 		6839 |     mdns_daemon_t *mdnsDaemon;
	if I disable #define UA_ENABLE_DISCOVERY_MULTICAST, I get the following link errors:

	/usr/bin/ld: open62541.o: in function `decodeNetworkMessage':
	/home/pi/OPCProject/open62541.c:48440: undefined reference to `verifyAndDecryptNetworkMessage'
	/usr/bin/ld: open62541.o: in function `UA_ReaderGroup_decodeAndProcessRT':
	/home/pi/OPCProject/open62541.c:53229: undefined reference to `verifyAndDecryptNetworkMessage'
	/usr/bin/ld: open62541.o: in function `setDefaultConfig':
	/home/pi/OPCProject/open62541.c:106474: undefined reference to `UA_ConnectionManager_new_MQTT'
	collect2: error: ld returned 1 exit status
	make: *** [Makefile:31: myNewClient] Error 1

dhugoexe, 2025-04-09:
Simillar issue. I'm trying to amalgamate open62541 with MQTT SubPub in order to run it on an ESP target. 

```bash
mkdir build && cd build

cmake -G "Ninja" -DCMAKE_TOOLCHAIN_FILE="C:\...\esp-idf\tools\cmake\toolchain-esp32p4.cmake" -DUA_ARCHITECTURE=freertosLWIP 
-DUA_ENABLE_AMALGAMATION=ON 
-DUA_NAMESPACE_ZERO=REDUCED 
-DCMAKE_BUILD_TYPE=MinSizeRel 
-DUA_ENABLE_PUBSUB=ON 
-DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON 
-DUA_ENABLE_PUBSUB_FILE_CONFIG=ON
-DUA_ENABLE_MQTT=ON ..
```

where my toolchain file is: 

```bash
# esp32_toolchain.cmake

# Tell CMake we are cross-compiling for an embedded target.
set(CMAKE_SYSTEM_NAME "Generic")
set(CMAKE_SYSTEM_PROCESSOR "riscv32")

# Set the paths for the ESP32 toolchain
set(CMAKE_C_COMPILER "C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe")
set(CMAKE_CXX_COMPILER "C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe")

set(CMAKE_AR "C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-ar.exe")
set(CMAKE_RANLIB "C:/Espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-ranlib.exe")

# Force CMake to search for libraries and includes only in the provided sysroot (if applicable)
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)

# Override default linker flags to avoid Windows-specific options.
set(CMAKE_EXE_LINKER_FLAGS_INIT "")
set(CMAKE_EXE_LINKER_FLAGS "")
set(CMAKE_SHARED_LINKER_FLAGS_INIT "")
set(CMAKE_SHARED_LINKER_FLAGS "")

set(CMAKE_TRY_COMPILE_TARGET_TYPE "STATIC_LIBRARY")
```
The couple of `open62541.c/.h` are successfully generated. But after that, when I build my code in order to flash on target, got very similars errors from ninja (not cmake but I think it was no difference).

### What I've tryed:
- add an `mqtt_adapter.c/h` to resolve undefined references (cause mqtt client isn't included)
- add and remove some UA Flags (but never UA_ENABLE_AMALGAMATION)

jackybek, 2025-04-09:
does your method work?