open62541/open62541 #7204: Inconsistency with UA_ENABLE_XML_ENCODING

RolfKal, 2025-04-07:
## Description
An undefined UA_ENABLE_XML_ENCODING is disabling declaration of UA_EncodeXmlOptions, but that type is used unconditionally in namespace0_generated.c

## Background Information / Reproduction Steps

After updating to master HEAD, running CMake to update the Visual Studio project and compiling, I get "undefined symbols opts in namespace0_generated.c" errors. The reason is that the namespace0_generated.c file uses the UA_DecodeXmlOptions unconditionally, but that declaration is in types.h and only available when UA_ENABLE_XML_ENCODING is defined.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master, latest
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

RolfKal, 2025-04-09:
If I understand it correctly, the new namespace_generated logic makes UA_ENABLE_XML_ENCODINGG support mandatory and not anymore an option that can be disabled.

o6-nbeier, 2026-10-20:
Thank you for the report. The build with differing encoding settings was fixed in #7293, so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier