open62541/open62541 #7211: Undefine reference to UA_Server_enableAllPubSubComponents()

jackybek, 2025-04-09:
## Description
When following the example given in 
/examples/pubsub/tutorial_pubsub_connection.c, there is a call to UA_Server_enableAllPubSubComponents().
However, during compiling, it complain:

``
SV_main.c:(.text+0xcc4): undefined reference to `UA_Server_enableAllPubSubComponents'
/usr/bin/ld: SV_main.c:(.text+0xce8): undefined reference to `UA_Server_enableAllPubSubComponents'

``

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag): 1.4.11
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: debian bookworm 64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

DevTigerart, 2025-04-09:
Hi,
You don't have mentionned which options you gave to CMake.
Was UA_ENABLE_PUBSUB enabled ?
Best regards.

jackybek, 2025-04-09:
Yes .. it is enabled in the open62541.h

DevTigerart, 2025-04-09:
Can you tell me how do you compile the tutorial file ?
On my side I have compiled and installed the open62541 library with the flag UA_ENABLE_PUBSUB enabled.
Then when i compile the tutorial file with gcc -std=c99 tutorial_pubsub_connection.c -lopen62541, the compilation runs fine.

jackybek, 2025-04-09:
Ok I will try to compile tomorrow

jackybek, 2025-04-10:
I realised that all the examples in open62541/examples/pubsub/ do not use UA_Server_enableAllPubSubComponents().

jackybek, 2025-04-10:
Please advise if there is still a need to invoke UA_Server_enableAllPubSubComponents() for pubsub capabilities (according to the github examples).

closed 2025-04-10 by jackybek

DevTigerart, 2025-04-10:
I'm glad everything worked out for you.

To answer your question about my setup:
- Alma linux 9
- GCC 11.5.0 20240719
- CMake 3.26.5

I think either the library or open62541.c you use the first time was lacking of this function.

I have very little knowledge about the pubsub feature so I'm not sure I can help you about your last question.

Best regards.