open62541/open62541 #7219: `PublishResponse` received between `CreateMonitoredItemsRequest` and `CreateMonitoredItemsResponse` does not trigger the notification callbacks

carl-cimon, 2025-04-11:
## Description

I'm running into an issue with some OPC UA servers where some monitored items never get their value initialized. Looking into it further, it appears to be the same as https://github.com/open62541/open62541/issues/5226, but I'm creating a new issue as that one was closed.

Here is my understanding of the problem:
1. The client creates a subscription, and begins sending `PublishRequest`s periodically.
2. The client sends a `CreateMonitoredItemsRequest` , using `UA_Client_MonitoredItems_createDataChanges()`.
3. The client receives a `PublishResponse`, which includes notifications for some of the nodes in the `CreateMonitoredItemsRequest`.
4. The client attempts to process the `PublishResponse` using `processDataChangeNotification()`, but the handles for the monitored items is not yet known, so the notification is discarded.
5. The client receives the `CreateMonitoredItemsResponse` and is able to map the monitored item handles.
6. Any subsequent `PublishResponse`s can be successfully processed, but since the server thinks it already notified the client of some of the monitored item values, it does not send an update until the value is next changed, resulting in the value being uninitialized on the client.

When this occurs, the message

```
[2025-04-10 16:45:09.766 (UTC-0700)] warn/client	Could not process a notification with clienthandle 82 on subscription 87
```

is written for each of the affected nodes (logged in `processDataChangeNotification()`).

We have been using version `v1.3.8` for a while now, and I can confirm that the issue still occurs on `v1.4.11`.

## Background Information / Reproduction Steps

This does not occur every time, and is dependent on whether the server waits until sending the `CreateMonitoredItemsResponse` before including any notifications about them in a `PublishResponse`. I'd say there is about a 20% failure rate with this particular server (CODESYS OPC UA Server), and I have not encountered it before in the 3+ years I've been using OPC UA.

As was pointed out in https://github.com/open62541/open62541/issues/5226#issuecomment-1884458517, I'm not certain that the server is following the OPC UA spec. The `PublishRespsonse` is for a `PublishRequest` that was sent prior to the `CreateMonitoredItemsRequest`, so it feels wrong to me that it could include notifications for the newly monitored items, but I haven't found anything saying that it is not allowed. If this is against the spec, let me know and I'll notify the CODESYS team instead.

Used CMake options:

```bash
cmake ../open62541 ^
 -GNinja -DCMAKE_C_COMPILER="cl.exe" -DCMAKE_CXX_COMPILER="cl.exe" -DMSVC_TOOLSET_VERSION=141 ^
 -DCMAKE_BUILD_TYPE=Debug ^
 -DBUILD_SHARED_LIBS=OFF ^
 -DCMAKE_INSTALL_PREFIX="C:\BGD_ROOT\141\x64\debug\1.1.0\open62541" ^
 -DUA_ENABLE_ENCRYPTION=OPENSSL -DOPENSSL_USE_STATIC_LIBS=ON -DOPENSSL_ROOT_DIR="C:\BGD_ROOT\141\x64\debug\1.0.0\OpenSSL" ^
 -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_AMALGAMATION=OFF -DUA_NAMESPACE_ZERO=REDUCED
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): `v1.3.8`, `v1.4.11`
 - [x] Other OPC UA SDKs used (client or server): Possibly, I'm not sure what CODESYS uses for their server backend
 - [x] Operating system: Windows, Linux (does not appear to be OS dependent)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached: What file format is allowed?
 - [ ] Self-contained code example attached: Not sure this is needed/helpful, since it seems dependent on the server.
 - [ ] Critical issue: Probably not, but this is the second time it's being reported.

arohr, 2025-07-23:
Same issue here. We're testing v1.4.12 against a Siemens S7 PLC (Firmware v4.0)

carl-cimon, 2025-07-23:
We ended up just manually requesting a value read after any node was added as a monitored item, using `UA_Client_Service_read()`. This feels like it shouldn't be necessary (and I still think it is an issue from the server not following the spec), but it has been a reliable fallback.

arohr, 2026-02-21:
The manual reading after creating the subscription kind of works, but it's really a hack. We don't have this issue in applications where we use the C++ SDK from Unified Automation. This brings me to the conclusion that it's more likely that this is a bug in open62541 rather than in the UA Servers from Siemens/Codesys/Beckhoff. @jpfr could you have a look at it? Thank you very much.

jpfr, 2026-02-24:
Implemented in #7778.

@arohr @jbbjarnason  Please check if that solves the issues against the servers you are using.