open62541/open62541 #7228: Segfault in UA_Server_new() when using link-time optimization '/GL'

mapron, 2025-04-16:
## Description

We are using open62541 library and discovered quite consistent segfault in UA_Server_new() call when compiling with Whole Program Optimization (LTO, /GL flag for MSVC). 

We use following flags for all company 3rd party source for MSVC:
- /GL - enable link-time-optimization
- /O1 - average optimization
- /Oi - enabling intrinsics
(that set of flags applied to dozens of libraries, but we switched to passing them for open62541 recently and encountered current issue). 

Turns out, creating UA_Server will crash with this exact flags. 

call stack (running server_mainloop example):
```
1   memmove                         VCRUNTIME140                      
2   UA_Array_copy                   ua_types.c                   2143 
3   String_copy                     ua_types.c                   173  
4   LocalizedText_copy              ua_types.c                   1314 
5   UA_copy                         ua_types.c                   1549 
6   copyReferenceTypeNodeAttributes ua_nodes.c                   800  
7   UA_Node_setAttributes           ua_nodes.c                   866  
8   addNode_raw                     ua_services_nodemanagement.c 1072 
9   Operation_addNode_begin         ua_services_nodemanagement.c 1192 
10  Operation_addNode               ua_services_nodemanagement.c 1577 
11  addNode                         ua_services_nodemanagement.c 1637 
12  addReferenceTypeNode            ua_server_ns0.c              69   
13  createNS0_base                  ua_server_ns0.c              113  
14  initNS0                         ua_server_ns0.c              919  
15  UA_Server_init                  ua_server.c                  572  
16  UA_Server_newWithConfig         ua_server.c                  628  
17  UA_Server_new                   ua_config_default.c          81   
18  main                            server_mainloop.c            24   
19  __scrt_common_main_seh          exe_common.inl               288  
20  BaseThreadInitThunk             KERNEL32                          

```
Segfault in:  ua_types.c:2142 - UA_Array_copy()
```cpp
    if(type->pointerFree) {
        memcpy(*dst, src, type->memSize * size); // dst=garbade data, size = garbage data.
        return UA_STATUSCODE_GOOD;
    }
```

## Background Information / Reproduction Steps

version: reproduced on v1.4.11.1, and master 6ecfa9500761c681adb8cff9c5b92f5c76a38723
OS: Windows
Toolchain: MSVC 2022 32-bit, MSVC 2019 32-bit
Compiler version: 19.43.34809 for x86, 19.29.30158 for x86
WinSDK: 10.0.22621

Used CMake options:

```bash
cmake -G Ninja -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="/MD /O1 /Oi /GL /DNDEBUG"  -DUA_BUILD_EXAMPLES=ON -DUA_MSVC_FORCE_STATIC_CRT=OFF
```

run any example containing server, for example server_mainloop.c

## Workarounds
We understand non-standard flags is the first things to consider. However, consider the fact that two different major compiler version produce this issue (also on 2 different version of library).   
Removing any of /GL , /O1, /Oi stops issue from reproduction. Raising optimization to /O2 removes segfault as well.   
Another workaround is to remove __restrict attribute from config entirely:
```cpp
include/open62541/config.h.in:524
#ifdef _MSC_VER
// # define UA_RESTRICT __restrict
# define UA_RESTRICT
```
or remove it only from UA_ExtensionObject_setValueNoDelete:
```cpp
src/ua_types.c:838
void
UA_ExtensionObject_setValueNoDelete(UA_ExtensionObject *eo,
                                    void * /*UA_RESTRICT*/ p,
                                    const UA_DataType *type) {
```
Optimizer seems to be confused about restrict pointer passing and we suspect subtle restrict contract violation could be a core issue.

## Q&A
Q: Why not patching it for ourselves / just changing options?
A: We believe that code can contain some issues that can affect someone else someday.

Q: Why not report this to Microsoft bug tracker.
A: Removing __restrict to fix an issue is a very suspicious to prove that optimizer is wrong.

mapron, 2025-04-16:
Here are two screenshots from WinDbg session, 
No "__restrict" attr, correct codegen before ns0_addNode_raw call.
[image]

With "__restrict" attr, missing codegen before ns0_addNode_raw call.
[image]

jpfr, 2025-04-19:
The non-aliasing implied by restrict should be considered in the function that gets called. Not in the function that calls.

Try to remove the restrict keyword in the header / declaration. But keep it for the function implementation.

Does that change the generated code?

mapron, 2025-04-20:
> Does that change the generated code?

not really, it is the same. Still crash with restrict present in declaration.

 Not sure if your claim is still true for MSVC LTO implementation...