open62541/open62541 #7239: UA_String_vprintf seems inconsistent

RolfKal, 2025-04-25:
## Description

UA_String_vprintf() has functionality to work with to small input string and just return the number of required characters. But there seems to be some inconsistency.

## Background Information / Reproduction Steps

mp_vsnprintf() return -1 if either str->data or str->length is 0. This causes UA_String_vprintf() to return with UA_STATUSCODE_BADENCODINGERROR;
Later on if str->length is greater than 0 it fails with UA_STATUSCODE_BADENCODINGLIMITSEXCEEDED if the return value of mp_vsnprintf() is greater or equal to str->length.

There is no way UA_String_vprintf() ever gets beyond this point, so the allocation for a new string later on can never happen.

For this function to work as intended, I think the check for mp_vsnprintf() should be removed. But of course this was likely added for the many uses of mp_snprintf() in the library.

## Checklist

Please provide the following information:

 - [*] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [*] Operating system: Windows, but not relevant
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

RolfKal, 2025-04-26:
I see that #7231 seems to address this already in a way. However it removes the check completely. It may be useful to actually check for a NULL pointer with a non-zero length, as this will of course crash.

jpfr, 2025-04-27:
The updated code does not crash.
It will only count the length and not write anything when the length is zero.

RolfKal, 2025-04-28:
Yes if string->length is 0 it will count the characters without trying to format into the string->data element. What I meant is that if someone passes in a string that is formatted initialized as:

`string->length = something bigger than 0`
`string->data = NULL or invalid pointer`

Of course this is a clear caller error and there is always the question if a library should try to protect the caller from shooting in their own feet or just let them find out and feel the pain.