open62541/open62541 #7248: Async method calls bypass access control.

17steen, 2025-05-06:
## Description

If asynchronous execution is enabled on a method, calls to the method will by pass access control, which results in unauthorized method access.

Used CMake options:

```bash
-DUA_MULTITHREADING:STRING=100
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.15
 - [x] Other OPC UA SDKs used (client or server): c client library and uaexpert
 - [x] Operating system: ubuntu2204
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Self-contained code example attached
 - [x] Critical issue: Any user relying on access control that enabled async methods may experience unauthorized method execution.

Code example: this is a combination of the access_control example and the async_method example

getUserExecutable and getUserExecutableOnObject both always return false, the method should never execute.

But as tested on uaexpert: 
[image]
The method call succeeds.

```c
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <open62541/plugin/accesscontrol_default.h>
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

#include <open62541/client_config_default.h>

#ifndef WIN32
#include <pthread.h>
#define THREAD_HANDLE pthread_t
#define THREAD_CREATE(handle, callback) pthread_create(&handle, NULL, callback, NULL)
#define THREAD_JOIN(handle) pthread_join(handle, NULL)
#define THREAD_CALLBACK(name) static void * name(void *_)
#else
#include <windows.h>
#define THREAD_HANDLE HANDLE
#define THREAD_CREATE(handle, callback) { handle = CreateThread( NULL, 0, callback, NULL, 0, NULL); }
#define THREAD_JOIN(handle) WaitForSingleObject(handle, INFINITE)
#define THREAD_CALLBACK(name) static DWORD WINAPI name( LPVOID lpParam )
#endif

static UA_Server* global_server;
static volatile UA_Boolean running = true;

THREAD_CALLBACK(ThreadWorker) {
    while(running) {
        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
                    "Try to dequeue an async operation");
        const UA_AsyncOperationRequest* request = NULL;
        void *context = NULL;
        UA_AsyncOperationType type;
        if(UA_Server_getAsyncOperationNonBlocking(global_server, &type, &request, &context, NULL) == true) {
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "AsyncMethod_Testing: Got entry: OKAY");
            UA_CallMethodResult response = UA_Server_call(global_server, &request->callMethodRequest);
            UA_Server_setAsyncOperationResult(global_server, (UA_AsyncOperationResponse*)&response,
                                              context);
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "AsyncMethod_Testing: Call done: OKAY");
            UA_CallMethodResult_clear(&response);
        } else {
            /* not a good style, but done for simplicity :-) */
            sleep(5);
        }
    }
    return 0;
}

static UA_Boolean
getUserExecutable(UA_Server *server, UA_AccessControl *ac,
                   const UA_NodeId *sessionId, void *sessionContext,
                   const UA_NodeId *methodId, void *methodContext) {
    printf("Called getUserExecutable\n");
    return UA_FALSE;
}
static UA_Boolean getUserExecutableOnObject(UA_Server *server, UA_AccessControl *ac,
                                        const UA_NodeId *sessionId, void *sessionContext,
                                        const UA_NodeId *methodId, void *methodContext,
                                        const UA_NodeId *objectId, void *objectContext)
{
    printf("Called getUserExecutableOnObject\n");
    return UA_FALSE;
}

static UA_StatusCode
helloWorldMethodCallback(UA_Server *server,
                         const UA_NodeId *sessionId, void *sessionHandle,
                         const UA_NodeId *methodId, void *methodContext,
                         const UA_NodeId *objectId, void *objectContext,
                         size_t inputSize, const UA_Variant *input,
                         size_t outputSize, UA_Variant *output) {
    UA_String tmp = UA_STRING_ALLOC("Hello world!");
    UA_Variant_setScalarCopy(output, &tmp, &UA_TYPES[UA_TYPES_STRING]);
    UA_String_clear(&tmp);
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Hello World was called");
    return UA_STATUSCODE_GOOD;
}

static void
addHelloWorldMethod(UA_Server *server) {
    UA_Argument outputArgument;
    UA_Argument_init(&outputArgument);
    outputArgument.description = UA_LOCALIZEDTEXT("en-US", "A String");
    outputArgument.name = UA_STRING("MyOutput");
    outputArgument.dataType = UA_TYPES[UA_TYPES_STRING].typeId;
    outputArgument.valueRank = UA_VALUERANK_SCALAR;

    UA_MethodAttributes helloAttr = UA_MethodAttributes_default;
    helloAttr.description = UA_LOCALIZEDTEXT("en-US","Say `Hello World`");
    helloAttr.displayName = UA_LOCALIZEDTEXT("en-US","Hello World");
    helloAttr.executable = true;
    helloAttr.userExecutable = true;

    UA_NodeId method_node_id = UA_NODEID_NUMERIC(1, 62541);

    UA_Server_addMethodNode(server, method_node_id,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                            UA_QUALIFIEDNAME(1, "hello world"),
                            helloAttr, &helloWorldMethodCallback,
                            0, NULL, 1, &outputArgument, NULL, NULL);

    UA_Server_setMethodNodeAsync(server, method_node_id, UA_TRUE);
}

static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

static UA_UsernamePasswordLogin logins[2] = {
    {UA_STRING_STATIC("peter"), UA_STRING_STATIC("peter123")},
    {UA_STRING_STATIC("paula"), UA_STRING_STATIC("paula123")}
};

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    global_server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(global_server);
    UA_ServerConfig_setDefault(config);

    /* Disable anonymous logins, enable two user/password logins */
    config->accessControl.clear(&config->accessControl);
    UA_StatusCode retval = UA_AccessControl_default(config, true, NULL,
             &config->securityPolicies[config->securityPoliciesSize-1].policyUri, 2, logins);
    if(retval != UA_STATUSCODE_GOOD)
        goto cleanup;

    /* Set accessControl functions for nodeManagement */
    config->accessControl.getUserExecutable = getUserExecutable;
    config->accessControl.getUserExecutableOnObject = getUserExecutableOnObject;

    /* Start the Worker-Thread */
    THREAD_HANDLE hThread;
    THREAD_CREATE(hThread, ThreadWorker);

    addHelloWorldMethod(global_server);

    retval = UA_Server_run(global_server, &running);

    /* Shutdown the thread */
    THREAD_JOIN(hThread);

 cleanup:
    UA_Server_delete(global_server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

logs when the method is executed:

```
[2025-05-06 19:59:46.918 (UTC+0200)] info/server	Try to dequeue an async operation
[2025-05-06 19:59:46.918 (UTC+0200)] info/server	AsyncMethod_Testing: Got entry: OKAY
[2025-05-06 19:59:46.918 (UTC+0200)] info/server	Hello World was called
[2025-05-06 19:59:46.918 (UTC+0200)] info/server	AsyncMethod_Testing: Call done: OKAY
```

neojaw, 2025-05-07:
See #5858.

As far as I understand, `UA_Server_call` is actually *supposed* to bypass access control because it runs on the local admin session, so actually it's not really suitable for async methods (except for toy examples). As far as I know there is currently no way to pass the session of the caller.

jpfr, 2025-05-26:
I will check to make sure the async job is only created when the calling session has the required access rights.

For the next 1.5 release we can think to push the session information into the async job.