open62541/open62541 #7249: [Question] Which NodesetLoader should be used with open62541/1.4.11.1?

fischjo3SICKAG, 2025-05-07:
I plan to add open62541 version 1.4.11.1 to its Conan recipe at [Conan Center Index](https://github.com/conan-io/conan-center-index/tree/master/recipes/open62541) with an option for NodesetLoader.

To the open62541 maintainers: Is this acceptable for you?

If yes, please review the pull request at Conan Center Index for this. There is an open question regarding which tag or commit of the [open62541-nodeset-loader](https://github.com/open62541/open62541-nodeset-loader) repository should be used:

| Commit | Pros | Cons |
|--------|------|------|
| [nodesetLoader@a6ef450](https://github.com/open62541/open62541-nodeset-loader/tree/a6ef45054ff74e0d798fccc257e4646355dadb6a) | Original commit used by tag 1.4.11.1 | Encounters an assertion failure every time an XMLElement DataType is loaded. |
| [nodesetLoader@76e0994](https://github.com/open62541/open62541-nodeset-loader/tree/76e0994cff81d8c9d2eca9f7f6a9d14519d3e49e) | Works fine (at least on my machine) | Not the commit referenced by 1.4.11.1. |
| [nodesetLoader@fbedde3](https://github.com/open62541/open62541-nodeset-loader/tree/fbedde3f81c6c517ce0656661c13e46503b19342) | Does not compile due to include errors (at least on my machine) | Not the commit referenced by 1.4.11.1. |

Considering this, there are several options to proceed:
- Submit a bug report to the open62541 project to fix the include errors upstream and wait for the next release with the bug fix.
- Use commit 76e0994 for the Conan package.

To the maintainers: What is your opinion?

jpfr, 2025-05-26:
The nodeset loader is much better for master.
Can you wait a few weeks until the 1.5 release?

Further also the nodeset submodule needs to match. We test the noresetloader in the CI.
So it should work for 1.4 just fine…

fischjo3SICKAG, 2025-06-24:
@jpfr I noticed that the functions UA_NODEID and UA_GUID are used by nodesetloader, and these are only available [if UA_ENABLE_PARSING is enabled](https://github.com/open62541/open62541/blob/9f3e5f461ad5ea04cde0d8e13f34dcb1dfc33a5d/include/open62541/types.h#L365). This means that using nodesetloader also requires UA_ENABLE_PARSING. However, NodesetLoader uses these functions only in two places ([Value.c](https://github.com/open62541/open62541-nodeset-loader/blob/a6ef45054ff74e0d798fccc257e4646355dadb6a/backends/open62541/src/Value.c#L243) and [NodeSet.c](https://github.com/open62541/open62541-nodeset-loader/blob/a6ef45054ff74e0d798fccc257e4646355dadb6a/src/Nodeset.c#L524)). If you could remove these usages, the dependency on UA_ENABLE_PARSING could be eliminated.