open62541/open62541 #7257: Difference between Repeated Callbacks and MonitoredItems

jackybek, 2025-05-10:
Can i know what are the differences between a RepeatedCallbacks vs MonitoredItems?

Under what circumstances do i choose repeatedcallbacks over monitoreditems?

jackybek, 2025-05-10:
Is my understanding as follows correct ?

1. Monitoreditems are like interrupts. Whenever there is a change in the value of the item under monitoring, it will flag the server.

2. Callbacks are like loops. It is called every x milliseconds.  The smallest interval is 50 milliseconds.

If I want to reduce the callback intervals further to 10 milliseconds, how do I achieve this ?

jackybek, 2025-05-11:
I just checked the repo and realised that while monitoreditems can be created during runtime in the server, callbacks cannot. Is my understanding correct.

This means if my server nodes are dynamically created, only monitoreditems can be used for notification when changes to the node under monitoring are detected.  Correct ?

jpfr, 2025-05-26:
Repeated callbacks are local in the client or server. They are not an opc ua Concept.

MonitoredItems concern notifications from a server. These are either for datachanges or events.

closed 2025-05-26 by jpfr