open62541/open62541 #7264: Build pipeline fails on 1.4 branch

keba-estr, 2025-05-19:
Test PR (1.4.12 without relevant changes): #7263

The following tests FAILED:
	 21 - check_eventloop_eth (Failed)
	 68 - check_pubsub_connection_ethernet (Failed)

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4.12
 - [X] Critical issue

jpfr, 2025-05-20:
This is a fallout from the switch to Docker containers in the CI.
@andreasebner FYI

andreasebner, 2025-05-21:
We started backporting CI pipeline features from the master branch to enable more tests and reactivate the pipeline. Due to the deprecation of Ubuntu 20 runners on GitHub, we have switched to a Docker-based environment, which gives us more flexibility. See #7276

closed 2025-05-25 by andreasebner