open62541/open62541 #7271: UA_Server_getSessionById needed

m-obi, 2025-05-20:
Hey. I'm implementing for my server the method ChangePassword from UserManagement. In the Core Specs it's written that I have to check the encryption.
Therefore I need to get the current session/channel by the Id to check the securityMode.
Why is getSessionById not public?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master (v1.4.11)
 - [ ] Other OPC UA SDKs used (client or server):
 - [x] Operating system: Win10/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2025-05-20:
The idea is that you customize the AccessControl plugin.
There you get all the credentials (and session id) and then can attach your own metadata to the session.

m-obi, 2025-05-21:
Yes I'm doing this cause I'm including the Server/ServerConfiguration/UserManagement into my AccessControl plugin. I'm also using the sessionContext.
But since all the sessions are already saved by the server I thought I can just get the things I need and don't need to save it a second time.

Now I "found" UA_EndpointDescription *endpointDescription in activateSession. Then I have to attach endpointDescription->securityMode to my sessionContext or, I also set userIdentityToken as session attribute within activateSession, should I use userIdentityToken->encryptionAlgorithm (UA_UserNameIdentityToken)?

jpfr, 2025-05-21:
You can attach custom attributes to a session in a key-value list:

https://github.com/open62541/open62541/blob/7525d1227071513d3d3f88ca9aa4e3a38d591e1f/include/open62541/server.h#L511

You can reopen this if required.

closed 2025-05-21 by jpfr