open62541/open62541 #7304: BadDecodingError cause of PublicationDate

m-obi, 2025-06-04:
## Description

I got a BadDecodingError when loading my custom nodeset made by SiOME. The PublicationDate is "2025-05-22T16:33:44+02:00". When I changed this manually in the generated code to "2025-05-22T16:33:44Z", it was fine.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2025-06-04:
Do you mean the PublicationDate that is located within the Model or RequiredModel tag?
I adjusted this value in the DI nodeset for test purposes, but was unable to reproduce the error.

NoelGraf, 2025-06-12:
merged

closed 2025-06-12 by NoelGraf