open62541/open62541 #7336: [Pubsub] VLAN tag not set properly when using ETH-UADP

Bubeck-ISW, 2025-06-26:
## Description

Using the latest [master](https://github.com/open62541/open62541/tree/c06b52af6da89870cc3897e8752a99c6d5b47d84) branch, the VLAN tag is wrong bit-shifted in the function `setETHHeader()` located in the file `eventloop_posix_eth.c` at line [194](https://github.com/open62541/open62541/blob/c06b52af6da89870cc3897e8752a99c6d5b47d84/arch/posix/eventloop_posix_eth.c#L190-L194).

```
   /* Set the 802.1Q VLAN header */
    if(vid > 0 && vid != ETH_P_ALL) {
        *(UA_UInt16*)&buf[pos] = htons(0x8100);
        pos += 2;
        UA_UInt16 vlan = (UA_UInt16)((UA_UInt16)pcp + (((UA_UInt16)dei) << 3) + (vid << 4));
        *(UA_UInt16*)&buf[pos] = htons(vlan);
        pos += 2;
    }
```

## Background Information / Reproduction Steps

The VLAN tag calculation appears incorrect due to improper bit-shifting of the pcp, dei, and vid fields. According to the IEEE 802.1Q standard, the correct VLAN tag format should be:

- PCP (3 bits) → bits 13–15
- DEI (1 bit) → bit 12
- VID (12 bits) → bits 0–11

To fix this, line [194](https://github.com/open62541/open62541/blob/c06b52af6da89870cc3897e8752a99c6d5b47d84/arch/posix/eventloop_posix_eth.c#L194) could be replaced with the following corrected expression:

```
UA_UInt16 vlan = (UA_UInt16)((((((UA_UInt16)pcp) << 1) + ((UA_UInt16)dei)) << 12) + vid);
```

## Checklist

Please provide the following information:

- [x] open62541 Version (release number or git tag): latest [master](https://github.com/open62541/open62541/tree/c06b52af6da89870cc3897e8752a99c6d5b47d84) branch
 - [x] Operating system: WLS Ubuntu 22.04.5 LTS
 - [x] Critical issue

jpfr, 2025-07-20:
Good find.
I think you then need to shift pcp by 13 bits.
Please prepare a pull request. You can also tell us to go ahead with this instead.

Bubeck-ISW, 2025-07-22:
@jpfr, I’ve prepared a pull request addressing the VLAN tag issue. Looking forward to your review and feedback.