open62541/open62541 #7341: When the publishing interval is 3600 * 1000, the server did not send a publishing response

fangjj0709, 2025-06-30:
## Description

Hello,when I used UA Expert to connect to the server and create monitoring items with a publishing interval of 3600 * 1000 ms and a sampling interval of 100 ms, the client sent a normal publishing request, but after one hour, the server did not send a publishing response.When the sampling interval is small, the published response is sent on time as usual.Setting the maximum sampling interval on the server to 20000000 has no effect.

[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.6
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

fangjj0709, 2025-07-02:
When I set the publishing interval to two minutes, I found that there was a publishing response, but the monitoring item value on UAExpert's end was still NULL

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier