open62541/open62541 #7375: How to use CMake inside the Cmake-build-system of more complex embedded project?

NetoBF, 2025-07-25:
Hello, 

The search didn't give me any answer.

So my problem is, we have a more complex embedded firmware which is build up modular and useable by different projects with their wanted settings and configurations (which are done via configuration-headers), which is also set up via CMake. all 3rd party libraries natively support a smooth embedding into our CMake build-system, except open62541. 
the CMakeLists.txt contains a lot of harmfull stuff that will be done to the whole project. e.g. setting the CMAKE_C_FLAGS-variable globally, which is also used by other 3rd party tools. It's not done privately for just this library, which is of course possible to do so in cmake. 
So when I just include the open62541 directory with "add_subdirectory" out whole program won't build anymore. We also use our own mbedTLS library for our whole product but this somehow gets corrupted and throws errors as soon as we add the open62541.

I tried saving the c-flags before adding the subdirectory, set the compile flags set by open62541 private to the lib and restoring the flags afterwards for our project. Unfortunately this didn't solve the problem, so there are probably other settings done by this cmake that I can't understand.

Also it generates a lot of stuff and it also generates the config-header from CMake-settings which is an unfortunate situation since we mostly use headers for nearly every configuration inside our projects. Then stack should be located in our platform-module and the user should not need to set any cmake-variables in his project for using the platform. For us a project would define the config.h by themselves in their repo and pass it to our platform, not by setting cmake-defines. I would now need to replace the config-header that was generated...

I tried manually adding all the sources and building them, like we also do sometimes with other libraries, but since the library relies a lot on generated files, this is also no option, except I generate them before. 

The whole cmake-situation is pretty unfortunate for modular embedded firmwares since it modifies so much stuff for the whole project.

Wouldn't it make more sense to create the library when a function is called and keep nearly everything private? As well somehow enabling a path that users can add their own libraries e.g. for the mbedTLS if they already have one for their system? I didn't find that option. 

Maybe I am doing something completely wrong but all answers I found didn't help for that problem.

jpfr, 2025-07-26:
Try the amalgamation build, aka single-file distribution.
That gives you a single pair of .c/.h files.
Pretty easy to handle in an embedded build.

NetoBF, 2025-07-28:
Is this maintainable as well? we are currently using simply a subtree and we do not want to modify any file in the open62541 module. the amalgamation would need to be done outside the project, not inside its cmake.
My next approach would be pre-generating some files that are needed for a build and then create our own cmake which does the build. 
At least for now that seems to be promising. 
I think if I do it either way the question comes up:
Are the functionalities activatable/deactivable via the config-header? I saw that a lot of files have guards which activate the content only if the define in the config-header is set. Shouldn't matter then if amalgamation or the source-files are used, right?
And how does it then work to add our own mbedTLS? When I activate security it seems I can only select between some predefined libraries which are included in this repo. 
I would simply like to link our mbedTLS to the open62541 build.

V-Strassheim, 2025-08-01:
@jpfr , please correct me if I’m wrong, but I don't believe there's currently a simple way to use the system-provided mbedTLS with open62541.

I think, both options (amalgamated files or writing an internal pre-generated files with cmake) have drawbacks, especially regarding the integration of future releases.
In my opinion, the best mid-term solution would be to improve the current build system with an optional "clean integration mode."
I have a few ideas about this for our solution, but I haven't had time to work them out in detail yet.

Anyway, the good news is that this is an open-source project. If you work on an improvement now and incorporate it into the mainline, it will help you reduce the work required to include the next release.
Additionally, this would help others include open62541 it in their build systems and enable further improvement of the entire solution.

Would that be a feasible option?

NetoBF, 2025-08-01:
@V-Strassheim yes this would be a huge improvement.
We at least currently managed to integrate the open62541 into our project, but with a lot of issues. I copied all needed files for compilation into a separate folder, deleted files which must be replaced with my own implementation, because even some internal stack-files have hard-coded ifdefs and are defaulting to POSIX architecture, and then build the stack and build our eventloops later on and link them against the stack-library.
The approach to add a new architecture is also a bit complicated. It seems easy from the documentation but unfortunately it is not done by just creating your own "arch" folder and add the architecture there. There are some files that also need to be changed internally like the ua_config_default in the plugins, which only knows the predefined architectures at all and defaults to posix, which is not useable for us. the same for the eventloop.h. Also the stdout, just default prints with printf which can be harmfull for embedded systems, so I needed to reroute this to our own log. I mean there is somehow the logger, that can be used I guess. But I did not find any documentation on that. (Also the TCP-log uses macros which are not correctly layed out with ticlang-compilers. ticlang somehow does not like to end a string and beginning another string inside the printf. It then just also prints out all format specifiers without replacing the values) and then there is as well the debug_dump_pckgs which has a hardcoded printf as well.
Also the musl_inet_pton does predefined architecture ifdefs.

Those are just some issues. But I found neither the documentation for the new event-loop structure internally, nor did I understand the configuration well enough and it seems the part for adaption for the own architecture involves a lot of parts which are not that easily changeable from the outside, because of the current source-file structure. And then of course linking the mbedTLS and so on is an issue as well that wee need to have a look at. 

Changing the Cmake to make it more "integratable" into your own project is surely possible but will take some work, also structural work in the source-files to remove all those architecture-dependent defines. We are happy we just got it to compile and run... Without any internal documentation it's really hard sometimes to follow the code and to understand what's happening. 

Last time I worked with that stack was like 9 years ago with a different user-name here in gitHub. I even made some contributions (NetoLE) and there was the network-layer which seems to be completely removed now. Probably out of some reasons I did not get. maybe opc ua fx? The adaption for our system in that time seemed to be way easier. Now it looks like a mix of the old world and a new approach which is not implemented completely.

Anyway. We got it to run, we just cannot connect to a socket somehow. But that's what we are currently investigating. 
I can try to enforce the contribution to the stack a bit, but unfortunately as it goes inside companies: "it runs? well that's good, it's done". And, we need to create an MVP. It's not like I have the possibility currently to invest time on that. Neither company- nor private time, and I really don't like this situation, because you are totally right. 

But you know what. I will check if we can place someone with this project, trying to improve it. If we use this stack it also should be smoothly useable for us.

What would be needed in any case:
- clear separation of architecture-dependent implementations from internal stack-code (no more #ifdef ARCHITECTURE inside the internal stack-code)
- remove "printf" in code, and provide a redefinable user-implementation.
- being able to link own 3rd party stacks, like mbedTLS
- improving cmake so it's useable as a target/project, which does not pollute the super-projects cmake-build
- improving documentation on how event-loops are used, to improve user-experience (I saw there are some tutorial-doc files mentioned, but none of them exist. and the doxygen is helpful but it's hard to get an overall picture with that)
- optional: create a c++-interface which should easify adding own architectures and can ensure a bit more type-safety and adds inheritance e.g. for your own implementations from interfaces.