open62541/open62541 #7387: UA_Server_run_iterate() does not use the server lock, crashes when other methods such as UA_Server_run_shutdown() are called from another thread

RolfKal, 2025-08-05:
## Description
When creating a program with a separate background thread that calls UA_Server_run_iterate() repeatedly, it is dangerous to call other methods from the main thread that also operate on the eventloop in some way. This is especially true for UA_Server_run_shutdown() which does take the server lock but then operates on the eventloop and messes with it while the background thread is running.

## Background Information / Reproduction Steps

The recommended programming is to have another boolean flag that signals to the background thread to exit the loop and call UA_Server_run_shutdown() in the background thread after the loop. This avoids common crashes when trying to shutdown the background thread from main but I'm afraid it will still cause problems if other UA_Server methods are called from main and they in some way work with the eventloop.

Is there a good reason not to lock the server around the eventloop run call inside UA_Server_run_iterate()? This would also allow to call  UA_Server_run_shutdown() from main and also protect from any other potential parallel calls that somehow work with the eventloop.

## Checklist

Please provide the following information:

 - [*] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [*] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

RolfKal, 2025-09-20:
> Have you observed the crash or is this theoretical?

Yes I did see crashes when I just called UA_Server_run_shutdown() from my main thread, while the background thread was looping. I worked around it by passing a stop boolean inside the user data and set that from main and then exit the loop in the background thread and call UA_Server_run_shutdown() from there just before terminating that thread.

Works but is a bit cumbersome. And the tests and examples go even more hardcore by just using a global to terminate the thread. Easy and simple but not scalable to multiple sessions with each their own run thread.

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier