open62541/open62541 #7392: UAClient - The connection has timed out before it could be fully opened

jackybek, 2025-08-07:
[image]

I am trying to create a UAClient object in a spawn thread from main().c

However, i get this error: The connection has timed out before it could be fully opened.

Please refer to the diagram above to see the program flow.

Can someone advise what caused this error and how I can go about resolving this?
Thanks.

open62541 v1.4.12

sudo cmake -S .. -DCMAKE_C_COMPILER=/usr/local/gcc14.2.0/bin/gcc14.2.0 -DCMAKE_C_COMPILER_AR=/usr/local/gcc14.2.0/bin/gcc-ar14.2.0 -DCMAKE_C_COMPILER_RANLIB=/usr/local/gcc14.2.0/bin/gcc-ranlib14.2.0 -DOPENSSL_CRYPTO_LIBRARY=/usr/local/ssl/lib64/libcrypto.so -DOPENSSL_INCLUDE_DIR=/usr/local/ssl/include -DOPENSSL_SSL_LIBRARY=/usr/local/ssl/lib64/libssl.so -DUA_ARCHITECTURE=posix -DUA_DEBUG_FILE_LINE_INFO=ON -DUA_ENABLE_AMALGAMATION=OFF -DBUILD_SHARED_LIBS=OFF -DUA_ENABLE_DA=ON -DUA_ENABLE_DATATYPES_ALL=ON -DUA_ENABLE_DEBUG_SANITIZER=ON -DUA_ENABLE_DIAGNOSTICS=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_DISCOVERY_SEMAPHORE=ON -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_HISTORIZING=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_MQTT=ON -DUA_ENABLE_NODEMANAGEMENT=ON -DUA_ENABLE_NODESETLOADER=OFF -DUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS=ON -DUA_ENABLE_PARSING=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ENCRYPTION=ON -DUA_ENABLE_PUBSUB_FILE_CONFIG=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_TYPEDESCRIPTION=ON -DUA_ENABLE_XML_ENCODING=ON -DUA_FORCE_WERROR=ON -DUA_LOGLEVEL=500 -DUA_NAMESPACE_ZERO=FULL

jackybek, 2025-08-08:
These are the parameters I set in UA_ClientConfig object:

        UA_ClientConfig *config1 = UA_Client_getConfig(uaClient);
        UA_ClientConfig_setDefault(config1);

                config1->clientContext = NULL;

                // this will overwrite the initial logging level (TRACE, DEBUG, INFO, WARNING, ERROR, FATAL)
                // UA_LOGLEVEL_ERROR => least messages
                // UA_LOGLEVEL_TRACE => most messages
                const char* env_loglevel = getenv("SVR_LOGLEVEL");
                UA_Logger logger;

                if (env_loglevel == NULL)
                        logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_INFO);     //(UA_LOGLEVEL_FATAL);
                else
                {
                        switch (atoi(env_loglevel))
                        {
                                case 0 : logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_TRACE);
                                        break;
                                case 1 : logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_DEBUG);
                                        break;
                                case 2 : logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_INFO);
                                        break;
                                case 3 : logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_WARNING);
                                        break;
                                case 4 : logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_ERROR);
                                        break;
                                case 5 : logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_FATAL);
                                        break;
                                default : logger = UA_Log_Stdout_withLevel(UA_LOGLEVEL_FATAL);
                                        break;
                        }
                }
                logger.clear = config1->logging->clear;
                *config1->logging = logger;

                config1->timeout = 0;
                config1->localConnectionConfig = UA_ConnectionConfig_default;

                UA_ApplicationDescription_clear(&config1->clientDescription);
                config1->clientDescription.applicationUri = UA_STRING_ALLOC(APPLICATION_URI);
                config1->clientDescription.productUri = UA_STRING_ALLOC(PRODUCT_URI);
                config1->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en", APPLICATION_NAME);
                config1->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;

                config1->endpointUrl = UA_String_fromChars(endpointUrl);
     /**
     * Connection configuration
     * ~~~~~~~~~~~~~~~~~~~~~~~~
     */
                //config1->userIdentityToken = ???
                config1->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;  // require encryption
                config1->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep");
                config1->noSession = UA_FALSE;
                config1->noReconnect = UA_FALSE;        // true = dont reconnect, false = automatic reconnect
                config1->noNewSession = UA_FALSE;       // true = dont automatically create new session, false = automatic create new session

                // UA_EndpointDescription endpoint
                config1->endpoint.endpointUrl = UA_String_fromChars("opc.tcp://192.168.1.109:4840");
                        config1->endpoint.server.applicationUri = UA_STRING_ALLOC(APPLICATION_URI);
                        config1->endpoint.server.productUri = UA_STRING_ALLOC(PRODUCT_URI);
                        config1->endpoint.server.applicationName = UA_LOCALIZEDTEXT_ALLOC("en", APPLICATION_NAME);
                        config1->endpoint.server.applicationType = UA_USERTOKENTYPE_CERTIFICATE;        // UA_USERTOKENTYPE_USERNAME
                        config1->endpoint.server.gatewayServerUri = UA_STRING_ALLOC(APPLICATION_URI);
                        //config1->endpoint.server.discoveryProfileUri =
                //config1->endpoint.serverCertificate = // to be initialised in this function =>  SV_WebSocketsCommon.c : initialiseSecurityOpcUaClientCommonWebSockets()
                config1->endpoint.securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
                config1->endpoint.securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep");
[... 51 lines cut ...]
                else
                        UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "-------SV_WebSocketsHttps.c : config->securityPolicies : success");

                config1->securityPolicies->policyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep");

    /* Certificate Verification Plugin */
                //config1->certificateVerification ???

                //config1->authSecurityPoliciesSize = ???
                // UA_SecurityPolicy *authSecurityPolicies
                //config1->authSecurityPolicies.uri = ???
                //config1->authSecurityPolicyUri = ???
                #ifdef UA_ENABLE_SUBSCRIPTIONS
                        config1->outStandingPublishRequests = 10;
                        config1->subscriptionInactivityCallback = NULL;
                #endif

    /* Session config */
                config1->sessionName = UA_String_fromChars("SV_WebSocketHttpsClient");
                config1->sessionLocaleIds = NULL;
                config1->sessionLocaleIdsSize = 0;

                config1->endpoint.serverCertificate = certificate;

What other parameters did I missed such that this segmentation fault occurred?

jackybek, 2025-08-10:
the segmentation fault goes away after i made changes to the config.

jackybek, 2025-08-10:
Made the following additional changes in the initialization of uaClient object:

     config1->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;  // require encryption
                config1->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep");
                config1->securityPoliciesSize = 2;
                config1->securityPolicies[0].policyUri =  UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#None");
                config1->securityPolicies[1].policyUri =  UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep");  

But i now get segmentation fault at this point (ua_securechannel.c) : 

     securityPolicy->channelModule.newContext(securityPolicy, remoteCertificate, &channel->channelContext)

jackybek, 2025-08-10:
After making some changes to my code, the segmentation error does not occur but in ua_client_connect.c, in this function: verifyClientSecureChannelHeader(void *application, UA_SecureChannel *channel,
                                const UA_AsymmetricAlgorithmSecurityHeader *asymHeader),

1. the instruction UA_ByteString serverCert = getLeafCertificate(asymHeader->senderCertificate) returns a corrupted buffer.
2. as a result, the next line failed:

       if (serverCert.length > 0 &&
       !UA_String_equal(&serverCert, &client->endpoint.serverCertificate))

I checked the contents of client->endpoint.serverCertificate - it is the correct content (my OPCUA server certificate is used to pass into the uaClient object).

But serverCert in this case is corrupted.

What cause the getLeafCertificate(asymHeader ->senderCertificate) to be corrupted?  Is it has to do with the SSL certificate generation process?

jpfr, 2025-09-18:
If you see corrupted data, please run the program with Valgrind.
That will give you a lot of log information about the origin of the data.