open62541/open62541 #7393: Disabling pubsub / pubsub encryption leads to compile errors

HardwareToWebWizard, 2025-08-07:
## Description
Disabling UA_ENABLE_PUBSUB leads to various compiler errors. 
Enabling UA_ENABLE_PUBSUB  but disabling UA_ENABLE_PUBSUB_ENCRYPTION also leads to various compile errors.

## Background Information / Reproduction Steps
To reproduce the mentioned errors, use the open62541.h and open62541.c files from release v1.4.12 or v1.4.13 and compile and link them to a project.

Then comment out the mentioned defines (either deactivate pubsub completly or just disable the pubsub encryption)

It was mentioned that this should be fixed by this commit https://github.com/open62541/open62541/commit/5364fdd893416290af3826a06370fa621b9cd650, but it seems like the open62541.h and open62541.c files were not updated correctly?

Does it rely on further disabled/enabled defines to compile correctly or are the defines used the intended way?

### First disable pubsub completely:

// #define UA_ENABLE_PUBSUB
// #define UA_ENABLE_PUBSUB_ENCRYPTION

The errors I get once disabling pubsub completly:
open62541.c:239541:23: error: unknown type name 'UA_NetworkMessage'
239541 |                 const UA_NetworkMessage *networkMessage,
open62541.c:239545:44: error: request for member 'securityHeader' in something not a structure or union
239545 |     UA_Boolean isEncrypted = networkMessage->securityHeader.networkMessageEncrypted;
open62541.c:239572:23: error: unknown type name 'UA_NetworkMessage'
239572 |                 const UA_NetworkMessage *networkMessage,
open62541.c:239605:18: error: unknown type name 'UA_PubSubSecurityPolicy'; did you mean 'UA_SecurityPolicy'?
239605 |                  UA_PubSubSecurityPolicy *securityPolicy) {
       |                  ^~~~~~~~~~~~~~~~~~~~~~~
       |                  UA_SecurityPolicy
open62541.c: In function 'UA_DataSetMessage_encodeJson_internal'
open62541.c:239730:11: error: request for member 'header' in something not a structure or union
239730 |     if(src->header.dataSetMessageSequenceNrEnabled) {

It seems like that those files
    /**** amalgamated original file "/src/pubsub/ua_pubsub_security.c" ****/ 
    /**** amalgamated original file "/src/pubsub/ua_pubsub_networkmessage_json.c" ****/

do not check if UA_ENABLE_PUBSUB / UA_ENABLE_PUBSUB_ENCRYPTION is set or not.

### Try to compile with pubsub but without encryption

If #define UA_ENABLE_PUBSUB is set but // #define UA_ENABLE_PUBSUB_ENCRYPTION is not set, the following compile errors occur:

open62541.c:239605:18: error: unknown type name 'UA_PubSubSecurityPolicy'; did you mean 'UA_SecurityPolicy'?
239605 |                  UA_PubSubSecurityPolicy *securityPolicy) {
       |                  ^~~~~~~~~~~~~~~~~~~~~~~
       |                  UA_SecurityPolicy

Is seems like that this file 
    /**** amalgamated original file "/src/pubsub/ua_pubsub_security.c" ****/
does not check wether UA_ENABLE_PUBSUB_ENCRYPTION is set or not.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4.12 also tested with v1.4.13
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

JianbinLiu-CFLab, 2025-08-08:
Hi @HardwareToWebWizard,

Thanks for reporting this in detail — just to clarify a few points:

Did you use the `open62541.c` and `open62541.h` files **directly from the GitHub release** (v1.4.12 or v1.4.13), or did you generate them yourself using CMake with custom options like:

```bash
cmake .. \
  -DUA_ENABLE_PUBSUB=OFF \
  -DUA_ENABLE_PUBSUB_ENCRYPTION=OFF \
  -DUA_ENABLE_AMALGAMATION=ON
```

**Note: The macro UA_ENABLE_PUBSUB_ENCRYPTION is no longer supported in current versions — PubSub encryption depends on UA_ENABLE_ENCRYPTION instead.)**。I guess what you actually want is:

```bash
cmake .. \
  -DUA_ENABLE_PUBSUB=OFF \
  -DUA_ENABLE_ENCRYPTION=OFF \
  -DUA_ENABLE_AMALGAMATION=ON
```

If you’re using the release version, note that the amalgamated file was built with PubSub (and encryption) enabled by default.
Simply commenting out the macros in open62541.h will not remove the related code, because some PubSub source files (like ua_pubsub_security.c) are not protected with #ifdef UA_ENABLE_PUBSUB, and will always be compiled.

Let me know how you obtained the `.c/.h` files — if you didn't generate them yourself, I can help walk you through the correct process.

Best regards,
Jianbin

jackybek, 2025-08-10:
try this method. It worked for me.

1. version : 1.4.12
2. source : git clone https://github.com/open62541/open62541.git --branch v1.4.12 -c advice.detachedHead=FALSE
2.1 cd open62541
2.2 sudo git submodule update --init --recursive
3. During make, errors related to MQTT (almagamation=ON), but open62541.h is produced (along with open62541.c)
4. Next, set: almagamation=OFF, discovery_multicast=ON, make = passed
5. libopen62541.a is produced
6. use libopen62541.a and open62541.h in application Makefile and application source codes (but don't use open62541.c in Makefile)

closed 2025-08-14 by HardwareToWebWizard