open62541/open62541 #7414: Error message prints an 'Empty string'  in processERRResponse() function

jackybek, 2025-08-16:
## Description

## Background Information / Reproduction Steps
in this function 

     processERRResponse(UA_Client *client, const UA_ByteString *chunk) {

     ...
     UA_LOG_ERROR_CHANNEL(client->config.logging, &client->channel,
                         "Received an ERR response with StatusCode %s and the following "
                         "reason: %.*s", UA_StatusCode_name(errMessage.error),
                         (int)errMessage.reason.length, errMessage.reason.data);

     ...
     }

I have a UA_Server running with Opccert109.pem certificate.

When i try to create a UA_Client (websockets) in a spawn thread to connect to the UA_Server, i get this error message:

     [2025-08-16 08:39:50.881 (UTC+0800)] error/channel      TCP 15  | SC 0  | Received an ERR response with StatusCode BadSecurityPolicyRejected and the following reason:   
     [2025-08-16 08:39:50.882 (UTC+0800)] info/client        Client Status: ChannelState: Closing, SessionState: Closed, ConnectStatus: BadSecurityPolicyRejected

As you can see, the 'reason' part of the first error message is blank.

So does it mean that there is a bug in the code since there is really no error ?

I already 
1.  encrypted the UA_Client with the server certificate
2.  setAUthenticationUsername returns success, 
3.  the server also returns 'The Aes128Sha256RsaOaep security policy channel with openssl is created'
4. loaded the trustlist Opccert109.pem since the spawn thread (UA_Client) is also running on the same server instance.

But I still get the error.
From the error message it seemed that there is nothing wrong with the certificates and loading of it, just that the Policy is rejected.
What caused the policy to be rejected?

I need the 'reason' but it is missing from the error message.
Please advise.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X ] open62541 Version (release number or git tag): 1.4.13
 - [X ] Other OPC UA SDKs used (client or server): opens62541 client and server
 - [X ] Operating system: debian bookworm
 - [X ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X ] Critical issue

jackybek, 2025-08-16:
This is what i got when i inspected the contents of servercert and client->endpoint.serverCertificate variables in ua_client_connect.c

printf("serverCert is \n %s \n", serverCert.data);
printf("client->endpoint.serverCertificate is \n %s \n", client->endpoint.serverCertificate.data);

serverCert is
 0▒B0▒*▒*▒H▒▒-Wd▒▒A▒H▒0

client->endpoint.serverCertificate is
 -----BEGIN CERTIFICATE-----
MIIHQjCCBSqgAwIBAgIUZcnIpiEWf9zxFC1XZIzlAkHYSKUwDQYJKoZIhvcNAQEN
BQAwgaQxCzAJBgNVBAYTAlNHMRIwEAYDVQQIDAlTaW5nYXBvcmUxEjAQBgNVBAcM
CVNpbmdhcG9yZTEWMBQGA1UECgwNVmlydHVhbCBTa2llczELMAkGA1UECwwCSVQx
IzAhBgNVBAMMGm9wYzEwOS52aXJ0dWFsc2tpZXMuY29tLnNnMSMwIQYJKoZIhvcN
AQkBFhRqYWNreTgxMTAwQHlhaG9vLmNvbTAeFw0yNTA4MTAxMzA5NTZaFw0zNTA4
MDgxMzA5NTZaMIGkMQswCQYDVQQGEwJTRzESMBAGA1UECAwJU2luZ2Fwb3JlMRIw
EAYDVQQHDAlTaW5nYXBvcmUxFjAUBgNVBAoMDVZpcnR1YWwgU2tpZXMxCzAJBgNV
BAsMAklUMSMwIQYDVQQDDBpvcGMxMDkudmlydHVhbHNraWVzLmNvbS5zZzEjMCEG
CSqGSIb3DQEJARYUamFja3k4MTEwMEB5YWhvby5jb20wggIiMA0GCSqGSIb3DQEB
AQUAA4ICDwAwggIKAoICAQC8YYySSv8DJAdFs4XOjPqN+Ua1g2iDyZ7jooT9jT73
Grw3uUoG641Nv7iK/S5XatFro9pwdV2qK1zgE1oJx4phq3WhsQluW9w/d8noWvvI
gm7rX/AjGc2Zh2Z5bPFBOZJWssuSC7Rq2ZadzmPh1p/l9MeI46jPPBXljJ4KeF2v
z3BvrNGUr07dfipZJ25WEcnD5fMLnnmOljhepONb4UXD5GhXmI79TCnJU8n8guD4
LwLdy9XNlNZvdl4oiSIYymw6yoF60GCFEJJvXmBW7+3caEimMzmk74DSn/bi6Xvg
SmJD6z9Yg2OvRFzMvCXCBKr/OEtHXAKF9vxH+QKo4s0x8twJoXUT/7RtYeXlHeit
6w3tTwhMdppWCpZiTfx/h+cSvt13jIrKRCbbOOzocgnHhUH1rf5l3x6ZAHYeYDx3
wLILflSf6PNv76sv8fg/b8rsSLFrYn08PfN7ezble7OO1Vmkl7cvP8f+S+k0Brht
kkY/QCDDJdXQtZFgLrFlAE/qm4n13ifXMdmC3IgQmoRDKuMNuOBv/qaXU7y697uF
lD2r+B7pOTA54ZpkkPLbb2H/pFZWFVA04S0ikWNDodp3xU9OYsT32tMAy+JHeD7k
fl5r48c7vWBqz8YfNiJbKDDq2816kP+pTvQ4Tbth4mXF27KE3HEl92Q9DW78ecfo
3wIDAQABo4IBaDCCAWQwgc4GA1UdIwSBxjCBw6GBqqSBpzCBpDELMAkGA1UEBhMC
U0cxEjAQBgNVBAgMCVNpbmdhcG9yZTESMBAGA1UEBwwJU2luZ2Fwb3JlMRYwFAYD
VQQKDA1WaXJ0dWFsIFNraWVzMQswCQYDVQQLDAJJVDEjMCEGA1UEAwwab3BjMTA5
LnZpcnR1YWxza2llcy5jb20uc2cxIzAhBgkqhkiG9w0BCQEWFGphY2t5ODExMDBA
eWFob28uY29tghRlycimIRZ/3PEULVdkjOUCQdhIpTAdBgNVHQ4EFgQUlU+e6gMB
AAFhDTVwaI6IqRL/TlQwCQYDVR0TBAIwADALBgNVHQ8EBAMCAvwwHQYDVR0lBBYw
FAYIKwYBBQUHAwIGCCsGAQUFBwMBMDsGA1UdEQQ0MDKGHnVybjpvcGMxMDkudmly
dHVhbHNraWVzLmNvbS5zZ4cEwKgBbYIKT1BDU3ZyLTEwOTANBgkqhkiG9w0BAQ0F
AAOCAgEATBIZ3T8KmIYU2gBkXiiMISZZtEJaJDpvWIIJAiQS8w+EQaFq/h0rRxho
HelDQrGcu1KLq4/buX8Oc5JEtrE7SDulO31C6fkmaqZOvP1aKD55P+zQrnuuFC1a
jNn1/U0gXFXk7tO/Cw5BN54ThvQkST/U00j7XT7UiLOkbwPDf94+azSrIDsQzbcv
Zhzu4hyMk75VL0NVbijRPoSwBtRIk1bf6R3q/lAWlzSF2U1EUc32ktsSlmo1WjeN
i9/ynw8YNcL/GCG8GTCkWYGey5ETXFOIy/OdpePCoyrUTF0dLM9i0tWRGTT09zSK
lLruNq4NcufJeMOhb/xTfaWpwsDbWQfuA+kTYirRDXDAG8YYQJwN0zPQx7T3LnQ3
FDyFwEbPNVsKUlZ5xssNLNw9tfYIKZ8+y/OByCJ9qhdgLvutLgKhDHzMJ4oh+hUi
KvMdb2blYZ9y+VP2RqVlrGa2VM97WW/WoFGoFk4eq8zmAd2Cw0VTCpLq+Anfm/5c
7NjzK/Iq0J9uSFt0z0VFWrv8d2WLsKioHAZN1g6YZXQ2aVHtblSyZTAdJTDciJie
8sksHbAXkMq+s8QbdT8ucQmuZulPorUx28XBLk4wuRov4mgaLRgUbb381lufrKtg
aAW8m/gfJPHguz9P0/Y3H9PHCJXWMP8Aro4QUEx7S5Q7vQOdq24=
-----END CERTIFICATE-----

As you can see, the contents of servercert is corrupted.

jpfr, 2025-09-18:
You have found the function that generates the error.
It shows that the server cert needs to be identical to the endpoint description.
That should point you to the problem in your config.

My advise is to start with a working example from our repository and make small modifications until the problem appears.
That way we have a minimal example that is easier to work with.