open62541/open62541 #7427: UA_BuildInfo_clear() segmentation fault occurred when running in pre-empt RT kernel : Linux OPCEtherCAT-111 6.12.11 #1 SMP PREEMPT_RT Mon Jan 27 02:47:25 AEDT 2025 aarch64 GNU/Linux

jackybek, 2025-08-24:
## Description
Running an existing working executable in linux bookworm has no such issues:

1. UA_BuildInfo_clear() - segmentation fault
Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3362
3362    ./malloc/malloc.c: No such file or directory.
(gdb) trace
Tracepoint 1 at 0x7ff6982a00: file ./malloc/malloc.c, line 3362.
(gdb) backtrace
#0  __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3362
#1  0x0000000000486554 in UA_Array_delete (p=0xc, size=13594304, type=0xc883e0 <UA_TYPES+144>) at /home/pi/open62541/src/ua_types.c:2082
#2  0x00000000004807d0 in String_clear (s=0xd067a8, _=0x0) at /home/pi/open62541/src/ua_types.c:181
#3  0x00000000004835c0 in DiagnosticInfo_clear (p=0xd06790, _=0xc88a10 <UA_TYPES+1728>) at /home/pi/open62541/src/ua_types.c:1173
#4  0x0000000000483eb0 in clearStructure (p=0xd06780, type=0xc8abd0 <UA_TYPES+10368>) at /home/pi/open62541/src/ua_types.c:1388
#5  0x0000000000483eb0 in clearStructure (p=0xd06780, type=0xc8dfd8 <UA_TYPES+23688>) at /home/pi/open62541/src/ua_types.c:1388
#6  0x0000000000484134 in UA_clear (p=0xd06780, type=0xc8dfd8 <UA_TYPES+23688>) at /home/pi/open62541/src/ua_types.c:1475
#7  0x0000000000463e0c in UA_BuildInfo_clear (p=0xd06780) at /home/pi/Downloads/airgardServer-109/open62541.h:39804
#8  configureServer (uaServer=uaServer@entry=0xd06770, userid=userid@entry=0x7ffffff1e0 "jackybek",
    password=password@entry=0xcb30e0 <hashpassword> "8ebe744e41fa3494536e9648093ab4f4ae156071eb54274c1dbf4c320c2023e0") at SV_Configure.c:116
#9  0x0000000000412bf8 in main (argc=1, argv=0x7ffffff558) at SV_main.c:464

2. UA_Server_addObjectNode() errors

[2025-08-25 01:42:49.499 (UTC+1000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | ua_services_nodemanagement.c : AddNode: UA_Node_setAttributes() failed
[2025-08-25 01:42:49.499 (UTC+1000)] info/session       TCP 0   | SC 0  | Session "Administrator"       | AddNode: Node could not create a node with error code BadNodeAttributesInvalid
[2025-08-25 01:42:49.499 (UTC+1000)] error/userland     Error creating object node : Root Node AirGardSensor : BadNodeAttributesInvalid

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.4.12-dirty
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X ] Operating system: Pre-empt RT linux bookworm 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X ] Self-contained code example attached
        ("--------SV_CreateNodes.c : Begin the process of instantiation OPC UA Tree structure"));

        //reference to namespace of the server : initialised in SV_StartOPCUAServer.c
        //UA_Int16 ns_MKS = UA_Server_addNamespace(uaServer, "virtualskies.com.sg/MKS/");
        UA_Server_getNamespaceByName(uaServer, UA_STRING("virtualskies.com.sg/MKS/"), &namespaceIndex);
        //UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "New Namespace added with Nr. %d \n", ns_MKS);

        // Add new object called Airgard Sensor
        UA_NodeId r1_airgardroot_Id;  /* get the nodeid assigned by the server */
        //UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "---UA_NodeId = %d\n", r1_airgardroot_Id);

        UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
        retval = UA_Server_addObjectNode(uaServer, UA_NODEID_STRING(namespaceIndex, sNameSpace),
            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),               // parentNodeId
            UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),                // referenceTypeId original UA_NS0ID_ORGANIZES
            UA_QUALIFIEDNAME(namespaceIndex, "AirgardTcpStreams"),              // browseName UA_QUALIFIEDNAME(1, "AirGard"),
            UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),              // typeDefinition this refers to the Object Type identifier
            oAttr, NULL, &r1_airgardroot_Id);                           // attr, nodeContext, outNewNodeId assigned by the server
        if (retval != UA_STATUSCODE_GOOD)
        {
           UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "Error creating object node : Root Node AirGardSensor : %s", UA_StatusCode_name(retval));
           exit(UA_FALSE);
        }
        //#ifdef DEBUG
        else
            UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "Root Node created : Airgard %d", r1_airgardroot_Id.identifier.numeric);                // OK

 - [X ] Critical issue

jackybek, 2025-08-25:
I updated the version of open62541 to v1.4.13 but the errors still happened.

jpfr, 2025-09-18:
Probably you double-free memory.
Or memory was allocated on the stack and cannot be free’d.
Check if you put any static data into the build info structure during your configuration phase.

The valgrind tool is a great help to that end. Check it out!