open62541/open62541 #7434: Bug when generating Nonce with SecPol ECC_NistP256. Connection loss

ChrisCarmonaIgeteam, 2025-09-03:
## Description
Hi,

Currently I'm working on the use of ECC with open62541. I generated myself a client with open62541 that opens a secure channel with Security Policy NistP256 and a session with user/password authentication. The connection is achive successfuly and some operations such as browse request and read request are perform properly.  The issue here is that when the channel needs to refresh security token the connection is closed without givin the client any info. 

I tried to debug it myself and I found that whenever we try to process a UA_SECURITYTOKENREQUESTTYPE_ISSUE the lenght of the output bytestream is different of UA_SECURITYPOLICY_ECCNISTP256_NONCE_LENGTH_BYTES has you could see in the method UA_Sym_EccNistP256_generateNonce. In contrast, for request of type UA_SECURITYTOKENREQUESTTYPE_RENEW the lenght of the out matches UA_SECURITYPOLICY_ECCNISTP256_NONCE_LENGTH_BYTES and it calls this other function instead:

UA_OpenSSL_ECC_NISTP256_GenerateKey 

The return of this function is UA_STATUSCODE_BADINTERNALERROR. 

```
[2025-09-03 09:59:30.842 (UTC+0000)] info/security	The EccNistP256 security policy channel with openssl is created.
[2025-09-03 09:59:30.897 (UTC+0000)] warn/channel	ua_services_Securechannel.c (26) request->requestType 0
[2025-09-03 09:59:30.897 (UTC+0000)] warn/channel	ua_services_Securechannel.c (73)
[2025-09-03 09:59:30.897 (UTC+0000)] warn/channel	ua_services_Securechannel.c (90)
[2025-09-03 09:59:30.898 (UTC+0000)] info/channel	TCP 13	| SC 2	| Generating new local nonce
[2025-09-03 09:59:30.898 (UTC+0000)] info/channel	TCP 13	| SC 2	| NonceLength correct
[2025-09-03 09:59:30.916 (UTC+0000)] info/security	secPol_eccnistp256 (347)
[2025-09-03 09:59:30.917 (UTC+0000)] info/channel	TCP 13	| SC 2	| statuscode: 0x0
[2025-09-03 09:59:30.917 (UTC+0000)] warn/channel	ua_services_Securechannel.c (94)
[2025-09-03 09:59:30.917 (UTC+0000)] info/channel	TCP 13	| SC 2	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#ECC_nistP256 and a revised lifetime of 60.0s
[2025-09-03 09:59:31.547 (UTC+0000)] info/security	secPol_eccnistp256 (349)
[2025-09-03 09:59:31.705 (UTC+0000)] info/session	TCP 13	| SC 2	| Session "ns=1;g=b9d3fc50-df27-23bb-10b1-9b6918015055"	| Session created
[2025-09-03 09:59:32.267 (UTC+0000)] info/security	The EccNistP256 security policy channel with openssl is created.
[2025-09-03 09:59:32.268 (UTC+0000)] info/security	The EccNistP256 security policy channel with openssl is deleted.
[2025-09-03 09:59:32.268 (UTC+0000)] info/security	secPol_eccnistp256 (349)
[2025-09-03 09:59:32.268 (UTC+0000)] info/session	TCP 13	| SC 2	| Session "ns=1;g=b9d3fc50-df27-23bb-10b1-9b6918015055"	| ActivateSession: Session activated with ClientUserId "guest"
[2025-09-03 10:00:16.703 (UTC+0000)] warn/userland	No certificate store configured. Accepting the certificate.
[2025-09-03 10:00:16.737 (UTC+0000)] warn/channel	ua_services_Securechannel.c (26) request->requestType 1
[2025-09-03 10:00:16.738 (UTC+0000)] warn/channel	ua_services_Securechannel.c (73)
[2025-09-03 10:00:16.738 (UTC+0000)] warn/channel	ua_services_Securechannel.c (90)
[2025-09-03 10:00:16.738 (UTC+0000)] info/channel	TCP 13	| SC 2	| Generating new local nonce
[2025-09-03 10:00:16.739 (UTC+0000)] info/channel	TCP 13	| SC 2	| NonceLength correct
[2025-09-03 10:00:16.739 (UTC+0000)] info/security	secPol_eccnistp256 (344) status: 0x80020000
[2025-09-03 10:00:16.740 (UTC+0000)] info/channel	TCP 13	| SC 2	| statuscode: 0x80010000
[2025-09-03 10:00:16.740 (UTC+0000)] warn/channel	TCP 13	| SC 2	| Could not open a SecureChannel. Closing the connection.
[2025-09-03 10:00:16.741 (UTC+0000)] info/channel	TCP 13	| SC 2	| Processing the message failed. Channel already closed with StatusCode BadUnexpectedError. 
[2025-09-03 10:00:16.742 (UTC+0000)] warn/channel	TCP 13	| SC 2	| Processing the message failed with error BadUnexpectedError
[2025-09-03 10:00:17.252 (UTC+0000)] info/security	The EccNistP256 security policy channel with openssl is deleted.
[2025-09-03 10:00:17.252 (UTC+0000)] info/channel	TCP 0	| SC 0	| SecureChannel closed
[2025-09-03 10:00:17.253 (UTC+0000)] info/network	TCP 13	| Socket closed
[2025-09-03 10:00:17.254 (UTC+0000)] info/network	TCP 12	| Connection opened from "192.168.1.100" via the server socket 10
[2025-09-03 10:00:17.254 (UTC+0000)] info/channel	TCP 12	| SC 3	| SecureChannel created
[2025-09-03 10:00:18.784 (UTC+0000)] info/channel	TCP 0	| SC 0	| SecureChannel closed
[2025-09-03 10:00:18.785 (UTC+0000)] info/network	TCP 12	| Socket closed
```

as you can in the next picture, the moment disconnection happens is when that openSecureChannelRequest is sent.

[image]

I check other security policies and they differenciate dependin on the length of the Nonce. Could yo please tell me why is this perform? 

Is it just me or has somebody else experience this connection last when updating security token with NistP256? 

Could you help me fixing this issue so i can maintain the channel properly open ? 

## Background Information / Reproduction Steps

The lib is crosscompile for an embeded platform of 32 bits. The server is run on that platform while the client is on a Windows PC.  

The version used in both platforms is 1.4.10.  
The OS a propietary Linux OS.
The encryption provider is OpenSSL 

Used CMake options:

BUILD_SHARED_LIBS ON
UA_ENABLE_ENCRYPTION "OPENSSL"

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 1.4.10
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Linux
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [X] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Thanks in advance for any help you could give me! If there is any info missing please tell me

dejavugoose664, 2025-09-03:
This should be the fix. [5f13b56ef272.zip](https://github.com/user-attachments/files/22121338/5f13b56ef272.zip)

ChrisCarmonaIgeteam, 2025-09-10:
Second Update:

I saw what is the problem here. when trying to renew the securitytoken of the channel the reference of the channel context from the policy context is lost. See these structures: 

´´´
typedef struct {
    EVP_PKEY *localPrivateKey;
    EVP_PKEY *csrLocalPrivateKey;
    UA_ByteString localCertThumbprint;
    const UA_Logger *logger;
    UA_ApplicationType applicationType;
    struct _Channel_Context_EccNistP256 *channelContext;
} Policy_Context_EccNistP256;

typedef struct _Channel_Context_EccNistP256 {
    EVP_PKEY      *localEphemeralKeyPair;
    UA_ByteString localSymSigningKey;
    UA_ByteString localSymEncryptingKey;
    UA_ByteString localSymIv;
    UA_ByteString remoteSymSigningKey;
    UA_ByteString remoteSymEncryptingKey;
    UA_ByteString remoteSymIv;

    Policy_Context_EccNistP256 *policyContext;
    UA_ByteString remoteCertificate;
    X509          *remoteCertificateX509; /* X509 */
} Channel_Context_EccNistP256;

´´´

channelContext and policyContext are pointers that allow each other to access the other structure. When renew token is happening channelContext of policy->policyContext is null and thus, it brings up an error.  Here some traces where you can see that the pointer is lost:

´´´
[2025-09-10 13:55:45.594 (UTC+0000)] warn/channel	ua_services_Securechannel.c (26) reqType: 1 channelContext: 0xAA33DA28 policyContext: 0xB54F8
[2025-09-10 13:55:45.595 (UTC+0000)] warn/channel	ua_services_Securechannel.c (75)
[2025-09-10 13:55:45.595 (UTC+0000)] warn/channel	ua_services_Securechannel.c (92) ChannelContext 0xAA33DA28
[2025-09-10 13:55:45.595 (UTC+0000)] info/channel	TCP 13	| SC 2	| Generating new local nonce
[2025-09-10 13:55:45.595 (UTC+0000)] info/channel	TCP 13	| SC 2	| NonceLength correct channelCon: 0xAA33DA28 secPolCon: 0xB54F8  http://opcfoundation.org/UA/SecurityPolicy#ECC_nistP256
[2025-09-10 13:55:45.596 (UTC+0000)] info/channel	TCP 13	| SC 2	| sp->symmetricModule.generateNonce 0xAE058D27
[2025-09-10 13:55:45.596 (UTC+0000)] info/security	secPol_eccnistp256 (352) policyContext: 0x000b54f8 pctx: 0x000b54f8
[2025-09-10 13:55:45.596 (UTC+0000)] info/security	secPol_eccnistp256 (354) pctx->channelContext: 0x0, pChannelContextPointer: 0x0 
[2025-09-10 13:55:45.596 (UTC+0000)] info/security	secPol_eccnistp256 (361)
[2025-09-10 13:55:45.598 (UTC+0000)] info/channel	TCP 13	| SC 2	| statuscode: 0x80010000
[2025-09-10 13:55:45.598 (UTC+0000)] debug/channel	TCP 13	| SC 2	| Renewing SecureChannel failed
´´´
pctx->channelContext should be channelContext: 0xAA33DA28 but it is lost. PolicyContext is been initiated but then is not filled correctly. I'm looking for some workaround since generateNonce interface is already defined.